360Controller/360Controller #195: Xbox One controller doesn't work in Counter-Strike: Global Offensive

RaymanDK, 2016-01-05:
As the title says. The game expects a 360 controller. Reason why Microsoft on Windows made it so it pretends to be a 360 controller or shows up as one. You get the idea.

Continue with the amazing work. Really appreciate it, thank you.

FranticRain, 2016-01-19:
Does the game work with 360 controllers on Mac? Do you have any way of testing that? I know on Windows, you need to load a config file using the console. Look into that and see if it helps.

FranticRain, 2016-05-07:
CS:GO does work with this controller. Make sure you check your in-game controller settings.

closed 2016-05-07 by FranticRain

githubneal, 2016-11-19:
has anyone actually used it successfully on a MAC?? I haven't seen any evidence of this at all online. Not a single person.

MaddTheSane, 2016-11-19:
You can try configuring the controller via Steam’s Big Picture Mode.