360Controller/360Controller #196: Saitek AV8R-02 entry

poorlymac, 2016-01-09:
Can I please get the Saitek AV8R-02 added to 360Controller/Info.plist

                <key>SaitekAV8R02</key>
                <dict>
                        <key>CFBundleIdentifier</key>
                        <string>com.mice.driver.Xbox360Controller</string>
                        <key>IOCFPlugInTypes</key>
                        <dict>
                                <key>F4545CE5-BF5B-11D6-A4BB-0003933E3E3E</key>
                                <string>360Controller.kext/Contents/PlugIns/Feedback360.plugin</string>
                        </dict>
                        <key>IOClass</key>
                        <string>Xbox360Peripheral</string>
                        <key>IOKitDebug</key>
                        <integer>65535</integer>
                        <key>IOProviderClass</key>
                        <string>IOUSBDevice</string>
                        <key>idProduct</key>
                        <integer>52009</integer>
                        <key>idVendor</key>
                        <integer>1848</integer>
                </dict>

I have tested this on El Capitan with 015_beta3 by setting boot-args  to kext-dev-mode=1 and rebooting into recovery and csrutil disable and it works for me.

FYI only, I also tried compiling myself but El Capitan needs the old USB, I also tried the elCapitan version of the code but got the following:
Jan  9 17:45:19 iMac27 sudo[700]: schaappa : TTY=ttys000 ; PWD=/Users/schaappa ; USER=root ; COMMAND=/sbin/kextload /Library/Extensions/360Controller.kext
Jan  9 17:45:19 iMac27 com.apple.kextd[55]: kext-dev-mode allowing invalid signature -67050 0xFFFFFFFFFFFEFA16 for kext "/Library/Extensions/360Controller.kext"
Jan  9 17:45:20 iMac27 com.apple.kextd[55]: kext signature failure override allowing invalid signature -67050 0xFFFFFFFFFFFEFA16 for kext "/Library/Extensions/360Controller.kext"
Jan  9 17:45:20 iMac27 kernel[0]: kxld[com.mice.driver.Xbox360Controller]: The following symbols are unresolved for this kext:
Jan  9 17:45:20 iMac27 kernel[0]: kxld[com.mice.driver.Xbox360Controller]: 	__ZN11StandardUSB35getNextAssociatedDescriptorWithTypeEPKNS_23ConfigurationDescriptorEPKNS_10DescriptorES5_h
Jan  9 17:45:20 iMac27 kernel[0]: kxld[com.mice.driver.Xbox360Controller]: 	__ZN15IOUSBHostDevice9metaClassE
Jan  9 17:45:20 iMac27 kernel[0]: kxld[com.mice.driver.Xbox360Controller]: 	__ZN18IOUSBHostInterface9metaClassE
Jan  9 17:45:20 iMac27 kernel[0]: kxld[com.mice.driver.Xbox360Controller]: 	_utf8_encodestr
Jan  9 17:45:20 iMac27 kernel[0]: Can't load kext com.mice.driver.Xbox360Controller - link failed.
Jan  9 17:45:20 iMac27 kernel[0]: Failed to load executable for kext com.mice.driver.Xbox360Controller.
Jan  9 17:45:20 iMac27 kernel[0]: Kext com.mice.driver.Xbox360Controller failed to load (0xdc008016).
Jan  9 17:45:20 iMac27 kernel[0]: Failed to load kext com.mice.driver.Xbox360Controller (error 0xdc008016).
Jan  9 17:45:20 iMac27 com.apple.kextd[55]: Failed to load /Library/Extensions/360Controller.kext - (libkern/kext) link error.

closed 2016-01-14 by poorlymac

FranticRain, 2016-01-14:
Check out the new release I just built and let me know if the controller works.

poorlymac, 2016-01-16:
Confirmed working on another computer without kext-dev-mode and csrutil with 0.15_beta4, thanks again!