360Controller/360Controller #198: (beta 4) Pretend 360 messes button mapping

RodrigoCard, 2016-01-14:
I just quickly tested and noticed that the default button mappings for xone controllers are wrong when the new "pretend 360" box is checked.
* Reattaching the controller doesnt fix the problem
* reset remapping wont work too.

FranticRain, 2016-01-14:
I tested this before I released it. Weird that it worked for me, but somehow broke for you. I'll check it out in a bit.

FranticRain, 2016-01-14:
It looks good to me. What are you testing with that sees a problem?

RodrigoCard, 2016-01-14:
just a single regular 2013 xbox one controller, no other USB devices connected except for a Microsoft mouse, if that matters.

I will do a clean install and try again, just to be sure.

FranticRain, 2016-01-15:
I tested with a 2013 controller and it looks perfectly fine to me.

RodrigoCard, 2016-01-16:
So, I uninstalled the driver, installed it cleanly. My results are now, somehow.... worse!
please check the pictures:

1 - First time I connected the controller after a a fresh install/fresh boot, wrong mappings:
[image]

2 - After clicking "start remapping" and "reset remapping" once, now the mappings are correct:
[image]

3 - "Pretend 360" checked, incorrect mappings (and different from 1 and 2), "reset remapping" trick doesnt make difference.
[image]

** Note that, in all pictures, I am holding only the ABXY buttons, at the same time. they should be 0123 indexes (not respectively) to match the expected behaviour since the first version of the driver **

The app I used for checking the button indexes is my own "Joystick Show", I sell it for a buck, but, whatever, get it for free here: https://dl.dropboxusercontent.com/u/14890209/Joystick%20Show.app.1.1.2.beta.zip

By the way, I am on Yosemite 10.10.5

FranticRain, 2016-01-16:
I'm on 10.11.2, and I can't recreate any of these problems with a 2013 controller. I have no idea what the issue is.

RodrigoCard, 2016-01-16:
So, in all cases, running 10.11.2 you get the correct mappings?

I will install 10.11.2 in another partition if that is the case, cant update my main environment yet.

d235j, 2016-01-16:
@RodrigoCard what version of Mac OS X are you running? I'd prefer that we don't force users to upgrade to the latest major release.

RodrigoCard, 2016-01-16:
@d235j 10.10.5  , latest yosemite. Yeah, I dont like upgrading too fast, stuff breaks.

FranticRain, 2016-01-16:
Correct. When I plug it in, it works by default. When I change to pretend 360 and then disconnect and reconnect the controller, it works just fine. I have no idea why the OS version would change anything. All of this stuff is super basic.

RodrigoCard, 2016-01-16:
@FranticRain right! I will try 10.11.2 in my mac.

Could anyone in 10.10.5 and a xbox one controller test that? 
@d235j @MaddTheSane @electric-monk

TyanColte, 2016-01-16:
@FranticRain I'm running latest El Cap and i'm getting the same issues @RodrigoCard is getting the same stuff i posted in the issue i created last night.

FranticRain, 2016-01-18:
@RodrigoCard Let me know if this fixes your problems. It does nothing for Elite controllers. https://drive.google.com/file/d/0B5PLCKoMd4zGNy15NFR1TTRxVkE/view?usp=sharing

larianstudios, 2016-01-18:
Same here 10.10.5 pretend button messes up things.

larianstudios, 2016-01-18:
@FranticRain Tested from drive:
Button messed up fixed for 'Pretend'
Bug: Pretend button state is not saved between disconnection
Bug: Clicking 'Pretend' does not switch modes more than once
And even more issues with 'Pretend'

RodrigoCard, 2016-01-18:
Installed 10.11 in another HDD and here are my results:

**Current Beta4 in El Capitan:**
- The same as @FranticRain , did not experience the same issues I reported, except for the first connection. chaging any settings avoids the issue.

**The new version from google drive ( tested in both El Capitan and Yosemite ):**
- *No mapping issues anymore, even in the first connection! Nice!* :D 
- Looks like "pretend" works only in once per connection, to disable you need to re-insert the controller. no big deal for now. 
- It is usable in Towerfall again in "Pretend" mode! :)

FranticRain, 2016-01-18:
Awesome. I'll keep working on it, and that should fix it. Basically, the driver now always re-organizes Xbox One packets to match 360 controller packets. Everything uses the 360 HID, and pretty much the only difference between the two is the PID.

FranticRain, 2016-01-19:
@RodrigoCard Any idea how we can do a better job of changing between Pretend360 and Normal? That PadDisconnect(), PadConnect() process is full of issues, as you can see with Joystick Show. I'm thinking, maybe disabling and then enabling the controller after Pretend360 is activated. Any ideas?

RodrigoCard, 2016-01-19:
well, I am not too much aware of some of the internals, maybe worth a shot...
But disabling and re-enabling the controller using the "enable driver" check box OR physically, seems to revert back to "non-pretend" state even if it is enabled.

FranticRain, 2016-01-25:
Resolved in Beta 5.

closed 2016-01-25 by FranticRain