360Controller/360Controller #200: Setting 'Pretend 360' Causes pref pane to crash. (Elite Controller)

TyanColte, 2016-01-16:
@FranticRain When setting pretend 360 in the pref pane on 2 occasions so far I experienced crashes the first time was when unplugging the controller after setting the "pretend 360" setting. The other time was just me setting the "pretend 360" setting with the controller plugged in. Also on mine, after setting the pretend 360 setting the controller doesn't work at all. No input whatsoever. Even after unplugging and replugging the controller I get no input. Here's the crash report i got on the second time the prefpane crashed. 

    Process:               System Preferences [568]
Path:                  /Applications/System Preferences.app/Contents/MacOS/System Preferences
Identifier:            com.apple.systempreferences
Version:               14.0 (14.0)
Build Info:            SystemPrefsApp-304001000000000~1
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           System Preferences [568]
User ID:               501

PlugIn Path:             /Library/PreferencePanes/Pref360Control.prefPane/Contents/MacOS/Pref360Control
PlugIn Identifier:       com.mice.driver.360Controller.Prefs
PlugIn Version:          1.0.0d15 (1.0.0d15)

Date/Time:             2016-01-15 21:23:57.635 -0600
OS Version:            Mac OS X 10.11.2 (15C50)
Report Version:        11
Anonymous UUID:        4FF3BFB4-4273-97DF-BF5A-948BF0381C32

Sleep/Wake UUID:       E9084A2F-260A-4DC8-AE90-0B3EEC823799

Time Awake Since Boot: 420 seconds
Time Since Wake:       280 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BREAKPOINT (SIGTRAP)
Exception Codes:       0x0000000000000002, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
com.mice.driver.360Controller.Prefs v.1.0.0d15 (XBox 360 Controllers)
objc[568]: GC: forcing GC OFF because OBJC_DISABLE_GC is set
Performing @selector(changeSetting:) from sender NSButton 0x7fc3ba65cb00
*** CFHash() called with NULL ***

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   com.apple.CoreFoundation      	0x00007fff919d07a9 CFHash + 297
1   com.apple.CoreFoundation      	0x00007fff919c91ca CFBasicHashFindBucket + 650
2   com.apple.CoreFoundation      	0x00007fff919c8eed CFDictionaryGetValue + 125
3   com.apple.CoreFoundation      	0x00007fff91a4974f -[CFPrefsPlistSource alreadylocked_setValue:forKey:] + 223
4   com.apple.CoreFoundation      	0x00007fff91a4963e -[CFPrefsSource setValue:forKey:] + 62
5   com.apple.CoreFoundation      	0x00007fff919fea1d +[CFPrefsSource withSourceForIdentifier:user:byHost:container:perform:] + 1037
6   com.apple.CoreFoundation      	0x00007fff91a495ad _CFPreferencesSetValueWithContainer + 301
7   com.mice.driver.360Controller.Prefs	0x0000000112bffa6e SetController + 73
8   com.mice.driver.360Controller.Prefs	0x0000000112bfdd94 -[Pref360ControlPref changeSetting:] + 4793
9   libsystem_trace.dylib         	0x00007fff92c4607a _os_activity_initiate + 75
10  com.apple.AppKit              	0x00007fff9d14951d -[NSApplication sendAction:to:from:] + 460
11  com.apple.AppKit              	0x00007fff9d15e486 -[NSControl sendAction:to:] + 86
12  com.apple.AppKit              	0x00007fff9d15e3b0 __26-[NSCell _sendActionFrom:]_block_invoke + 131
13  libsystem_trace.dylib         	0x00007fff92c4607a _os_activity_initiate + 75
14  com.apple.AppKit              	0x00007fff9d15e30d -[NSCell _sendActionFrom:] + 144
15  libsystem_trace.dylib         	0x00007fff92c4607a _os_activity_initiate + 75
16  com.apple.AppKit              	0x00007fff9d15c932 -[NSCell trackMouse:inRect:ofView:untilMouseUp:] + 2693
17  com.apple.AppKit              	0x00007fff9d1b1d58 -[NSButtonCell trackMouse:inRect:ofView:untilMouseUp:] + 744
18  com.apple.AppKit              	0x00007fff9d15b05c -[NSControl mouseDown:] + 669
19  com.apple.AppKit              	0x00007fff9d708d1d -[NSWindow _handleMouseDownEvent:isDelayedEvent:] + 6322
20  com.apple.AppKit              	0x00007fff9d709fad -[NSWindow _reallySendEvent:isDelayedEvent:] + 212
21  com.apple.AppKit              	0x00007fff9d062735 -[NSWindow sendEvent:] + 517
22  com.apple.systempreferences   	0x000000010de55660 0x10de4e000 + 30304
23  com.apple.AppKit              	0x00007fff9d05ee49 -[NSApplication sendEvent:] + 2540
24  com.apple.systempreferences   	0x000000010de54d0e 0x10de4e000 + 27918
25  com.apple.AppKit              	0x00007fff9cf9203a -[NSApplication run] + 796
26  com.apple.AppKit              	0x00007fff9cf14520 NSApplicationMain + 1176
27  libdyld.dylib                 	0x00007fff9798d5ad start + 1

[... 331 lines cut ...]
shared memory                     16.3M        8 
===========                     =======  ======= 
TOTAL                            457.3M      627 

Model: MacBookPro7,1, BootROM MBP71.0039.B0E, 2 processors, Intel Core 2 Duo, 2.4 GHz, 8 GB, SMC 1.62f7
Graphics: NVIDIA GeForce 320M, NVIDIA GeForce 320M, PCI, 256 MB
Memory Module: BANK 0/DIMM0, 4 GB, DDR3, 1067 MHz, 0x80CE, 0x4D34373142353237334348302D4346382020
Memory Module: BANK 1/DIMM0, 4 GB, DDR3, 1067 MHz, 0x85F7, 0x483634353155363446373036364700000000
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x8D), Broadcom BCM43xx 1.0 (5.106.98.100.24)
Bluetooth: Version 4.4.3f4 16616, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en1
Serial ATA Device: Crucial_CT250MX200SSD1, 250.06 GB
Serial ATA Device: MATSHITADVD-R   UJ-898
USB Device: USB 2.0 Bus
USB Device: Card Reader
USB Device: USB 2.0 Bus
USB Device: Built-in iSight
USB Device: USB Bus
USB Device: BRCM2046 Hub
USB Device: Bluetooth USB Host Controller
USB Device: IR Receiver
USB Device: Apple Internal Keyboard / Trackpad
USB Device: USB Bus
USB Device: Controller
Thunderbolt Bus:

FranticRain, 2016-01-16:
There shouldn't be anything different about that button when its for the Xbox One controller. But this is what happens when you let me touch the pref pane!

TyanColte, 2016-01-16:
@FranticRain lol, so update on the issue. I was playing around with it, and i was able to get the setting to stick without crashing and the elite controller to be recognized. However the controls were crazy as hell. Like that one time when we were in early days of getting the elite controller working.

FranticRain, 2016-01-16:
Can you recreate the issue with a normal Xbox One controller or is it elite only?

FranticRain, 2016-01-16:
Also make sure to back out and re-enter the pref pane if you aren't getting inputs. It really doesn't like it when you connect devices while it is open.

TyanColte, 2016-01-16:
@FranticRain I haven't tried, I'm out of town but I brought my 2015 one with me so i won't be able to test with the 2013 controller. And yea, i edited my second post so you may want to take a look.

FranticRain, 2016-01-18:
@TyanColte Let me know if this helps you at all. https://drive.google.com/file/d/0B5PLCKoMd4zGQTdNNXlfUGV6NHc/view?usp=sharing

robfrizzy, 2016-01-18:
I can comment on this because I am having the same problem as OP. I used the build you linked above and plugged in my elite controller. It recognizes it and everything works fine. I click "pretend to be a 360 controller" and it registers no input. I back out of the control panel and back in. When I do this I still had no input. Then I went to the drop down menu at the top and it listed my controller twice. Selecting the second controller makes all the inputs all crazy like before. If I uncheck the "pretend to be a 360 controller" box, then I again lose all input from the controller. Backing out of the control panel and back in still leaves me with two controllers to select in the dropdown menu, but the selecting the second controller seems to work normally. Disconnecting the controller and reconnecting it resets the controller selection and lists only one controller again which works normally. 
Also toggling the saved mappings switch on the controller doesn't seem to do anything.

So, to list what I've done:
1. Connect Elite controller
  -Works normally
2. Check "pretend to be 360 controller" box
  -No input
3. Back out and back in
  -No input
  -Duplicate entry in controller selection list
4. Select second entry in controller selection list
  -Crazy inputs
5. Uncheck "pretend to be 360 controller" box
  -No inputs
  -Still duplicate entry in controller selection list
6. Back out and back into pref pane
  -No inputs
  -Still duplicate Entry in controller selection list
7. Select second entry in controller selection list
  -Inputs work as intended
8. Disconnect and reconnect controller
  -Inputs work as intended
  -Duplicate entry in controller selection list gone.

Sorry for the long post but I hope this helps.

FranticRain, 2016-01-18:
@robfrizzy That's what is supposed, sort of. Ideally, after you check the box, you need to unplug the controller and then reconnect it. So can you plug it in, check the values are correct, toggle the pretend 360, then unplug the controller, reconnect it, and see if the bindings are right or not.

robfrizzy, 2016-01-18:
No dice. I plug in my controller and it works like intended and all the bindings seem correct. I click the pretend box and I lose all input. Unplugging the controller and plugging it back in just leaves me with "No Devices Found" If I back out of the pane and back in it show the controller connected but the inputs are going crazy. Unplugging it and plugging it back in again crashes System Preferences. When I open the pane again the inputs are still crazy. Unchecking the pretend box and checking it again loses all input. Unplugging and plugging it back in restores input and functions properly, but it's not in pretend mode.

FranticRain, 2016-01-18:
Okay, thats good. I'll try something out and then get a new build to you that should work a little better.

robfrizzy, 2016-01-18:
Ok, thanks. If you would like a log or something let me know.

FranticRain, 2016-01-19:
Okay, I made some big changes so everything might be broken. Let me know if this works. I still can't fix the dumb disconnect reconnect thing, so you'll still have to physically disconnect the controller and back out of the system preferences. https://drive.google.com/file/d/0B5PLCKoMd4zGcGZCV0V0UmtBQXc/view?usp=sharing

TyanColte, 2016-01-19:
@robfrizzy thanks for testing this, @FranticRain sorry I haven't been able to give feedback. I've been super busy the last few days I haven't even been able to whip out the MacBook to try this. Next time I get the chance I'll go full on and test all this stuff myself and report my results.

robfrizzy, 2016-01-19:
Thanks, I'll try this one out tonight.

FranticRain, 2016-01-21:
Any news on this?

robfrizzy, 2016-01-21:
Sorry, I forgot to check it out. I'll try it out tomorrow night and let you know how it goes.

robfrizzy, 2016-01-22:
I installed the linked build and it's still all wacky. Same symptoms as before. Also, it seems that although I've programed the controller to invert the right stick using the accessories app on the xbox my left one is showing up inverted as well.

FranticRain, 2016-01-22:
I missed a bunch of stuff, this *should* work better. https://drive.google.com/file/d/0B5PLCKoMd4zGMHpZMlFHTWk3Sjg/view?usp=sharing

robfrizzy, 2016-01-22:
I installed that build you just put up. I get no response at all now. No input from the sticks, triggers or buttons. The only one that works is the guide button. Switching to pretend mode doesn't do anything either. There's no crazy input now, but there's also no input at all.

FranticRain, 2016-01-22:
Okay, I added a bit of debug that should tell me what is going wrong. If this one still doesn't work, open up the Console from Applications/Utilities and look for a line like: `Xbox One - report->header.size: ` https://drive.google.com/file/d/0B5PLCKoMd4zGR2lIbEhWZmNrWGM/view?usp=sharing

robfrizzy, 2016-01-22:
I think you've fixed it now. It looks to be working as intended and pretend mode seems to be working fine as well. I'll play a game and test it a bit more extensively.

robfrizzy, 2016-01-22:
Wait, it looks like the left stick is unintentionally inverted.

FranticRain, 2016-01-22:
Make sure you haven't changed anything using the accessories app, or haven't accidentally changed a value in the preference pane. That code looks good to me.

robfrizzy, 2016-01-22:
I have inverted the right stick on the accessories app and it is inverted but so is the right stick. I could test it by disabling this in the accessories app.

FranticRain, 2016-01-22:
I'd appreciate if you did that. The normal Xbox One controller and the Elite controller both call the same function to handle sticks and it works just fine for normal Xbox One controllers. The Elite controller should be inverting the one stick on its end and then handing good data off to me.

robfrizzy, 2016-01-22:
Ok, so I saved a default configuration with no changes to buttons or the sticks to the second slot on the controller. When I connect the controller and have slot one activated (profile with the right stick inverted) both sticks are inverted. Switching to profile two (default) makes the right stick not inverted but the left stick is still inverted. The inversion options in the preference pane are still wacky as noted in my previous post.

FranticRain, 2016-01-22:
Can you toggle the left stick invert in the preference pane just in case its some renegade setting that improperly saved?

robfrizzy, 2016-01-22:
Toggling "invert y" on the left stick inverts the right stick.

FranticRain, 2016-01-22:
What in the world? I'll plug in a controller in a little bit and see if I messed up the inverts somehow. This makes no sense.

robfrizzy, 2016-01-22:
I made a quick recording to show you what I mean.
https://vid.me/e/Qu71

FranticRain, 2016-01-22:
Found the bug, this build should be perfect. https://drive.google.com/file/d/0B5PLCKoMd4zGM3RqUm04bUY2Y1U/view?usp=sharing

robfrizzy, 2016-01-22:
This one seems to do the trick. Everything seems to be working perfectly. I also only had to back out and back into the preference pane to have to controller show up in "pretend 360" mode. Thanks so much!

TyanColte, 2016-01-24:
@FranticRain I'm closing this. I believe you covered everything in my original issue post. Thanks @robfrizzy for being there to help test when I was to busy. Lets do a release off this one.

closed 2016-01-24 by TyanColte