360Controller/360Controller #201: Unknown controller

andhie, 2016-01-17:
I got a 360 controller that doesn't work with 0.15 Beta 4. Here are some details from the System Profiler

Info | Details
---------|----------
  Product ID: | 0x02a0
  Vendor ID: | 0x045e  (Microsoft Corporation)
  Version: | 1.00
  Serial Number: | 0694C614
  Speed: | Up to 12 Mb/sec
  Manufacturer: | Microsoft Corporation.
  Location ID: | 0xfa130000 / 5
  Current Available (mA):	| 1000
  Extra Operating Current (mA): | 0

FranticRain, 2016-01-17:
Install this version of the driver and let me know if it fixes the problem. Is this the Microsoft Big Button controller? https://drive.google.com/file/d/0B5PLCKoMd4zGVmxUQlFvbXN2Z3M/view?usp=sharing

andhie, 2016-01-17:
Tested with the given driver. it now works! thanks a lot!
i'm not too sure what do you mean by Big Button. I bought it off the internet, not too sure if its even genuine. but it looks like this. 
[image]

FranticRain, 2016-01-17:
Its definitely not genuine. Thats the product ID of the IR receiver for this thing. But I'm glad it works for you!

[image]

closed 2016-01-17 by FranticRain