360Controller/360Controller #204: HUGE ISSUE - System unusable ! Please HELP :'(

BloodyFoxie, 2016-01-19:
Hi ! 

I installed the 360Controller 0.14, buy using the .dmg
The installation went on correctly, asked for my password and for a reboot, which I did. 
Except it never booted again : 
I'm stuck with the stop sign saying the system can't access critical files.
I booted on my other disk and repaired the main system permissions and disk but it still can't boot on it.

I have no idea why it did that but my system is unusable and my last backup is from months ago.
Is there anyway I can delete/remove what the app did to the system ? 
I don't know what it's been modifying ?

Thank you so much for your help, I might have done something the wrong way during the install...

BloodyFoxie, 2016-01-19:
EDIT : solved by reseting my TRIM, still don't know what happened though !

d235j, 2016-01-19:
@BloodyFoxie:
I think this is unrelated to the driver. However, I'd be concerned about your SSD possibly failing soon.
I'd recommend backing up your data, and if this is a third-party SSD, updating the SSD firmware.

I'm closing this issue since it's unrelated to the driver.

closed 2016-01-19 by d235j

BloodyFoxie, 2016-01-19:
Thank you for your quick answer ! :)