360Controller/360Controller #205: Metallics Limited Edition GameStop wired 360 controller

jimbo1111, 2016-01-19:
Hi, I have one of these http://www.gamestop.com/xbox-360/accessories/metallics-limited-edition-by-gamestop-wired-controller-for-xbox-360/126483 and a retina iMac running El Capitan, and the driver seems not to recognize my controller.  It tells me "no devices found".  It didn't work on Yosemite either.  I'm on beta 5.  I read around and it seems that maybe since not many people have the controller, it needs to be manually added to the driver.  I have xcode 7.2 and wouldn't know what to do with it anyway.

Here's the information in system profiler when the controller is connected.

Metallics LE Xbox 360 Controller:

  Product ID:	0x0159
  Vendor ID:	0x0e6f
  Version:	1.00
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Performance Designed Products
  Location ID:	0x14100000 / 11
  Current Available (mA):	1000
  Extra Operating Current (mA):	0

FranticRain, 2016-01-20:
Boy PDP really knows how to pump out controllers. Quite frankly, its impressive. This build should work for you, let me know if it doesn't. https://drive.google.com/file/d/0B5PLCKoMd4zGaE8yODNqTkQ0dWs/view?usp=sharing

closed 2016-01-20 by FranticRain

jimbo1111, 2016-01-20:
Seems to be working, you rock!