360Controller/360Controller #206: Xbox One force feedback is broken

FranticRain, 2016-01-20:
In moving to El Cap, force feedback stopped working for Xbox One controllers. I'm not quite sure why.

@RodrigoCard - I know you're not running El Cap yet, does Xbox One force feedback still work for you?

larianstudios, 2016-01-20:
El Cap 10.11.2 stock 0.15.4 everything fine, tests and game

FranticRain, 2016-01-20:
I just tested with El Cap 10.11.3 after a full uninstall and then clean install of 0.15.4. Unfortunately its still not working. Maybe I messed it up somehow with all of the development builds. Wireless 360 rumble still works. What a weird issue.

RodrigoCard, 2016-01-21:
Well, I did not test a proper game on my new separated El Cap partition. but it worked on the preference pane. and it worked well on yosemite on both towerfall (pretend on) and prefpane.

By the way, if you tested only on prefpane, it stops working sometimes and I am not sure why.

FranticRain, 2016-01-21:
I tested both Fez and the Pref Pane. I'm doing a clean wipe of OS X just in case. I backed up my certificates this time though!

FranticRain, 2016-01-21:
Full reinstall did not fix the issue. I'll keep looking into it. This is the point where I would normally blame the controller, but it rumbles just fine on Windows. I don't get it.

larianstudios, 2016-01-21:
Does the FF lib initialise correctly for controller (i.e. kext ff plugin loads fine?)

FranticRain, 2016-01-21:
It has to because I have have a wireless 360 controller than rumbles and an Xbox One controller that doesn't rumble plugged in at the same time.

FranticRain, 2016-01-21:
I've debugged it down to the point where `SetReport` is getting called, and the send says it is successful, but I get nothing. I even added rumble packets to startup to see if that would help. Still nothing. I'm connected directly to my rMBP. I've tried both USB ports. I'll try a different cable when I get home, but I highly doubt that is it.

FranticRain, 2016-01-21:
Nope, change of cables did nothing. This is the weirdest bug.

larianstudios, 2016-01-21:
Can you boot into Yosemite?

FranticRain, 2016-01-22:
I have it on a separate partition, and I just tested it. Still no rumble, but on the upside, I finally got to see the incorrect binding bug!

jeshuamj2, 2016-01-23:
I can confirm the rumble does not work for me even in El Cap 10.11.2. I am running the latest update. P.S. How do you test rumble from the pref pane? Do you just select default for rumble options? If this is the case, rumble doesn't work from pref pane either.

FranticRain, 2016-01-23:
Rumble intensity should be directly proportional to how much you depress the trigger.

zero334, 2016-01-24:
Like @jeshuamj2, I can confirm that bug.
No rumble in the pref pane.
Using El Cap 10.11.2 with a brand new Xbox One controller.

FranticRain, 2016-01-24:
It looks like this: https://github.com/paroj/xpad/commit/3160825254467fcff5fc755489025e93f5cb6430 might be the trick, but I haven't been able to crack the code yet. It seems like a simple fix, but I guess not.

FranticRain, 2016-01-25:
I think I fixed it. If it hasn't worked for you, this should work, and if it currently does work for you, does this still work? You can find a compiled version of it over here: https://github.com/360Controller/360Controller/pull/208#issue-128294877

zero334, 2016-01-25:
For me it's working now with your compiled version.

FranticRain, 2016-01-25:
Resolved in Beta 5.

closed 2016-01-25 by FranticRain