360Controller/360Controller #207: Xbox One Wired + UE 4.10.2 = Buggy analog joystick

CleanCut, 2016-01-21:
Starting with the custom beta4 from @RodrigoCard at https://drive.google.com/file/d/0B5PLCKoMd4zGNy15NFR1TTRxVkE/view?usp=sharing

Setup:
- Mac Pro Late 2013 (black cylinder)
- OS X El Capitan 10.11.2
- Xbox One connected via USB wire
- Unreal Engine 4.10.2

Description
- Everything seems great in the prefs pane.  (The above build fixes the initially-everything-is-mapped-wrong-problem that I was having before)
- In UE I am using the `InputAction LEFT` blueprint node, which is supposed to send a `pressed` event once the joystick is moving left of center, and a `released` event once the joystick gets back near center.  (Emulating holding a keyboard arrow without the analog values).
- With a wired Xbox 360 controller, everything works great.  The events fire as expected.

Problem
- With the Xbox One controller connected via USB, the `pressed` event initially fires correctly, but as soon as the joystick stops moving (while holding all the way to the left, for example) the `released` event fires.  It's reeeally bad if you just slowly move the joystick, because then the `pressed` and `released` events flood in almost continuously.
- I tried checking the `Pretend to be an Xbox 360 Controller` box and nothing changed.

I don't know if this is an Unreal Engine issue or a 360Controller issue, but since it works perfectly with an Xbox 360 controller and the Xbox One driver seems to be under active development, I figured I'd start here.

FranticRain, 2016-01-21:
When activating the pretend 360 option, you have to unplug the controller and re-connect it to to actually do anything. I've been throwing around all sorts of builds on this driver recently so I don't know exactly what the code for that version you are using looks like. This is the most up to date version of my branch of the driver, here: https://drive.google.com/file/d/0B5PLCKoMd4zGcGZCV0V0UmtBQXc/view

At this point, the driver just changes its USB VID and PID for the pretend 360 option, so make sure to check this out. Beyond that, it seems like an Unreal issue because it seems to work just fine with games and the Pref-Pane as is.

CleanCut, 2016-01-23:
@FranticRain Sorry for the delay.  I installed the driver you linked to, rebooted, turned on the pretend 360 option, reconnected the controller, and nothing changed.  It still behaves badly as I described above. :-(

FranticRain, 2016-01-23:
No problem, I'm downloading UE4 and I'll give this a look in the next couple of days to see what the deal is.

FranticRain, 2016-01-23:
I'm struggling to re-create the bug, do you have a simple project I could test it with that has some sort of debug output?

CleanCut, 2016-01-23:
Okay, I swear the Xbox 360 controller was working, but I went all the way back through all the drivers I have tried all the way down to 0.15beta3 and it's acting the same way.  I read some more docs and it looks like it's much simpler to just use the axis values (float from [-1.0, 1.0] and just clamp them after a threshold to treat them like buttons.

So maybe this is behaving as intended?  Either way, I found a simple way to get what I need that doesn't involve changing the driver, so I'm going to go ahead and mark this closed.  Thanks for looking into it!

closed 2016-01-23 by CleanCut

FranticRain, 2016-01-23:
Awesome. I'm glad you found a fix.