360Controller/360Controller #209: Not recognising up and down dpad buttons openemu

brlawlor, 2016-01-24:
When trying to map my controller on openemu the up and down dpad buttons aren't recognised as input options.  These are the only buttons to have this issue and the fact that neither are recognised I dont think it's a controller issue

FranticRain, 2016-01-24:
OpenEmu uses DDHidLib which is one of the best controller libraries I've ever seen. It should recognize the d-pad buttons with button numbers: 12, 13, 14, 15 to Up, Down, Left and Right respectively. Could you check that up and down on the d-pad work in the preference pane? I've never known DDHidLib to ever have problems with any controller.

FranticRain, 2016-08-09:
Closing due to inactivity.

closed 2016-08-09 by FranticRain