360Controller/360Controller #210: make OSX channel/key names compatible with windows ones

increpare, 2016-01-24:
It's creating an industry of people doing OS-specific rebindings downstream.

Ok this isn't a serious request, but I've always wondered what the rationale is for giving everything different names...

windows:
http://wiki.unity3d.com/images/thumb/a/a7/X360Controller2.png/600px-X360Controller2.png

vs osx 360
http://wiki.unity3d.com/images/thumb/5/5f/Unity_360controller_mac_layout.png/600px-Unity_360controller_mac_layout.png

vs xbone 360
http://answers.unity3d.com/storage/temp/46370-xo-controller.jpg

FranticRain, 2016-01-24:
I'm not the most well informed with the history of this driver to help you with that, but I'm *fairly* certain that that Xbox One controller layout shouldn't be used anymore. To explain, I'll have to give you a bit of a history lesson.

That binding in that picture is the OLDBIND layout for my Xone-OSX repo. (You can see the documentation for this if you scroll to the bottom of this page: https://github.com/FranticRain/Xone-OSX/blob/master/README.md) That was the first binding I put out when just getting the driver rolling. Then, when I realized it had terrible compatibility with old games, I changed the layout to match the Tattiebogle driver, that's the mapping right above the OLDBIND mapping on that README.

Now, the funny thing about the OLDBIND is that the implementation of the driver at the time was virtually identical to the Tattiebogle driver. I had essentially pulled the source from the Tattiebogle website and altered it to send the correct init packet to the Xbox One controller. I had also created an alternate HID report descriptor. This was because the Xbox One controller has a different packet layout than the Xbox 360 controller, which I originally interpreted as a need for a new report descriptor. The newer binding changed how things worked though. It used this driver's report descriptor, but that also meant that the packets had to be altered. Internally, the driver accepts an Xbox One style packet from the controller, it then converts all of the contents of the packet to identically match a 360 style packet, and then sends it to the computer. You should start to smell something potentially fishy here.

While googling around for some documentation on third party controllers, I came across the 360Controller repository. I started toying with the implementation, and was able to implement Xbox One controller support. This implementation was the same as the new binding on the Xone-OSX repo. It altered Xbox One packets to match 360 packets. That is how it works today, with the exception of 0.15 beta 4 which was a misinterpretation of a request, ignore that implementation, it will soon be replaced with beta 5 which fixes the problem.

Now, I downloaded Unity so I could get you the proper mappings for the Xbox One controller, but Unity staunchly refuses to recognize my Xbox One controller's existence. Because you see, the problem isn't with this or any other driver. I believe the problem is Unity itself. As I said, the OLDBIND version of the driver reports buttons in its report descriptor that are IDENTICAL to those buttons listed on that image you posted. But the 360Controller repo follows an almost identical process that uses a similar report descriptor, but all of its numbers are incredibly high, in fact it starts at button 5 and goes up to button 19, but the report descriptor never mentions anything that high. Why is that? Why do two drivers using almost identical implementations have such drastic differences? 

That is why I believe that this driver isn't the problem, Unity is the problem. Just think about it. If you were going to make a game engine, wouldn't you handle this mapping situation within the engine, so each developer didn't have to come up with their own version of the code to fix this problem? It is just another example of Unity half-baking an implementation and expecting developers to fix it. Maybe it's just my general distaste for big game engines talking, but I staunchly refuse to believe that the driver is the problem here. Especially when you can use other gamepad libraries like DDHidLib that recognize the button numbers like they are reported in the HID descriptor.

If something here was unclear, let me know and I'll do my best to clear it up for you.

slime73, 2016-01-24:
Unity is well known for having poor gamepad support / implementations.

electric-monk, 2016-02-05:
Just to second both previous replies, the problem is indeed Unity. The HID descriptor I created labels all the buttons in a way that seemed sensible (all independent analog inputs are axis, all buttons are buttons, and ordered sensibly).

A few years ago I investigated this Unity issue, and it turns out it completely ignores the HID descriptor outside of "it's a button" or "it's an axis", so all index/etc. information is lost and it basically is just zipping through all the HID items and plopping them in in whatever order it fancies. Nothing we can really do on our side that isn't a hack that couldn't be relied, and I'm not even sure that would work.

There's also the fact that unity loses information on the triggers, which is also undesirable.

I'm closing this!

closed 2016-02-05 by electric-monk