360Controller/360Controller #212: (Beta 5) Wrong DPAD mappings (both xbox one and xbox 360) + useless button

RodrigoCard, 2016-01-26:
- There is an extra button that does nothing before the dpad button indexes, (residual from one's sync button? the sync buttons does nothing)
- This happens in both xbox 360 and xbox one controllers this time.
- The XBOX GUIDE (home?) button is 10 on xbox 360 and 11 on xbox one
- If you press any button AND home together, the useless button fires and gets stuck until you press another button.

In this pic I am holding the guide button on both controllers (I also pressed another one and released):
[image]

FranticRain, 2016-01-26:
Oh yeah, I know what that is. Just got sloppy and forgot to get rid of the extra bit I added to the report descriptor. I'll fix it real quick.

FranticRain, 2016-01-26:
Latest version of the release should fix this, check it out.

FranticRain, 2016-02-12:
Going to go ahead and close this since it has been resolved.

closed 2016-02-12 by FranticRain