360Controller/360Controller #215: controller only works within the driver setup

jimmyf78, 2016-01-28:
controller only works within the driver setup/ making show every button recognized and working fine..
I've tried using xbox ONE controller (0x14100000 / 2) in many games but the games doesn't know there is a controller connected (wired and all lit up) I've tried different cables,
On Yosemite 10.10.5
360ControllerInstall_0.15_beta5_unofficial

FranticRain, 2016-01-28:
1st. What games specifically are you having trouble with?
2nd. Have you tried going into the "Advanced" tab and checking the box that says: "Pretend to be a 360 Controller?"

jimmyf78, 2016-01-28:
1st. Lego Batman 3, Deadpool, Tombraider...
2nd. Yes, it doesn't work.
Any suggestions?

FranticRain, 2016-01-28:
Did you use any Xbox One controller drivers before 360Controller? For example, if you used Xone-OSX, make sure you run the uninstall.command. I'm downloading Tomb Raider right now to test stuff.

jimmyf78, 2016-01-28:
yes i did / Enabler Tool for OS X but it didn't work so i found a guide for "Xone-OSX" which linked me further to 360ControllerInstall_0.15_beta5_unofficial

i uninstalled Enabler Tool for OS X with "app-delete"...
should i reinstall the Enabler Tool for OS X and delete it some other way?

Whats the uninstall-command?

FranticRain, 2016-01-28:
The uninstall.command is just a tool I added to Xone-OSX to help user's remove the driver, which gets hidden in /Library/Extensions. But the controller enabler is just a .app, so deleting it should be fine. I'm not overly familiar with it, so I can't be sure. I'll check my Xbox One controller against Tomb Raider when I finish downloading it.

jimmyf78, 2016-01-28:
thanks, very much appreciated..

FranticRain, 2016-01-28:
Okay, so it works for me with Tomb Raider just fine. Make sure that "Pretend to be an Xbox 360 Controller" is checked. Make sure that the name of the controller in the pulldown menu is "Xbox One Wired Controller (Xbox 360) (Wired)." When you open Tomb Raider, can you go to the "Advanced" tab of the "Tomb Raider Options" window? There is a "Gamepad and wheel configuration" section, and there should be connected devices listed in there. Is there anything in there?

jimmyf78, 2016-01-28:
ok so it works with tomb raider... i installed it just now... but still not working with lego batman witch is a native microsoft game from what i can tell from within the game (hint: B,A,X,Y and color mappings)
what could be the problem?

jimmyf78, 2016-01-28:
controller for deadpool isn't working either- but that isn't a big deal for me - just the lego game

FranticRain, 2016-01-28:
LEGO® Batman™ 3: Beyond Gotham is a Feral ported game just like Tomb Raider. It should work exactly the same. As for Deadpool, as far as I can tell, its not officially ported for Mac. What method are you using the play it? At this point it seems very unlikely I will be able to solve your problem. We can't fix games on a case by case basis, thats not how a driver works. If it works with Tomb Raider, that means that the driver is functioning properly, and there isn't much I can do.

OriginalDAN, 2016-02-08:
Hi. I just got a mac, and wanted to play minecraft with controller. I see the controller working in the controller test section, however when I go into minecraft control settings it doesn't respond to any of the control pad inputs. Is there anything you can do to help me? Thanks

FranticRain, 2016-08-09:
Closing out unfixable game specific issues.

closed 2016-08-09 by FranticRain