360Controller/360Controller #216: Driver doesn't seem to work with OSX Lion 10.7.5

daddadhad, 2016-01-29:
I am on 10.7.5 and when I plugin my controller (I tested it on a friends mac with this driver and the controller worked fine so I know it's not the controller) the green circle lights up then goes blank and the driver menu says no device connected. I have reinstalled, tried different versions of this driver still the same result. My default boot is 64 bit but i can boot into a 32 bit (I have also tryed that, it didn't work).

daddadhad, 2016-01-29:
Also i am using an xbox 360 controller if that helps.

FranticRain, 2016-02-12:
You may get some help from #34 but we currently don't release the driver for anything older than 10.9. Apple changed the location of kexts in the file system and changed signing practices with that release, so unfortunately, you'll have to use an old version.

closed 2023-05-21 by arcturus-prime