360Controller/360Controller #217: NSNotificationCenter postNotification on controller connect/disconnect

qasim, 2016-01-29:
Could we broadcast `NSNotificationCenter` notifications for when controllers are connected/disconnected? I am trying to listen to this event but only see one way of doing it, which would be to duplicate the logic in this project.

larianstudios, 2016-01-30:
Why do you need this? You should use HID disconnect/connect callbacks

qasim, 2016-01-30:
This is more for the scenario of a developer making an application that listens for Xbox controller being plugged in. Is the driver in charge of listening for this or does it only do it when your in the preference panel? If its the latter, I may have to write my own HID manager instead since it wouldn't fit well with this project.

larianstudios, 2016-02-01:
You should consider using SDL and/or other libraries instead. HID should be used when you really need it.

qasim, 2016-02-02:
I will look into SDL, it may not be applicable for my needs though. Here's how I'm accomplishing it for anyone else who may stumble upon this: https://github.com/qasim/big-picture-launcher/blob/master/big-picture-launcher/XboxControllerListener.m.

Thanks, @larianstudios!

closed 2016-02-02 by qasim