360Controller/360Controller #218: problem

carldw, 2016-01-30:
the joy stick in  war thunder  for mac   it makes the camera roll  it keeps turn to the left and u cant play the game while its rollling the right joystick its not my controller  it works in other games and doesnt to it

FranticRain, 2016-01-31:
This is a known issue with Warthunder that the game developers must fix. We cannot fix it.

carldw, 2016-02-08:
oh ok

kudit, 2016-02-11:
Having a similar issue in GTA San Andreas.  It thinks the triggers are the right joystick so the only way to play is by holding both triggers down half way (obviously not a great solution).  Any ideas why this is or how to disable the triggers?  Why can't I map the triggers?

FranticRain, 2016-02-11:
`how to disable the triggers`
You don't want this. You'd still have the same problem and no way to solve it.

`Why can't I map the triggers?`
Because that's a really bad idea. How would you duplicate a joystick using triggers? Its something you may think you want, but after about 3 seconds you would realize how dumb it was.
As for mapping buttons, I have said this a million times: "Remapping triggers as buttons is not a great idea. You would have to go through a series of buttons to get to the one you wanted, and then precariously hold a trigger in just the right position for the perfect amount of time. Additionally, it creates so much space for user error. Like imagine a trigger with one of the buttons mapped to start. Then you would open the menu for the game every time you tried to get to a specific button. The entire remapping menu would be a complete mess."

You may have noticed that the buttons in that game are also all wrong. That is because Rockstar put absolutely 0 effort into making that thing work with controllers on Mac. Its a bug they have to fix, which they never will. We cannot fix it. Your best bet is to download something like ControllerMate and convert your controller inputs into keyboard inputs.

But, by all means, if you think this is the wrong answer, you're welcome to implement it yourself and submit a pull request. The joys of open source!

Gadgeteering, 2016-07-13:
I have reported this bug to Gaijin http://forum.warthunder.com/index.php?/topic/322786-xbox360-controller/

FranticRain, 2016-08-09:
Closing out unfixable game specific issues.

closed 2016-08-09 by FranticRain