360Controller/360Controller #220: Can't get the newest, or any driver to recognize my xbox 360 controller.

inasimplerhyme, 2016-01-31:
I recently purchased an Xbox 360 controller and 360 Receiver and can't get the driver to recognize the controller.  (It just says "No Devices Found").  I've searched through articles here and most of everything I've ready is WAY over my head.  But if there's anything I can try and do to give some information that will help solve the problem, I'll be glad to try and help.

I'm on Mac OS 10.10.5 on a 2013 Mac Pro.  Here's my products:

USB Receiver

http://www.amazon.com/gp/product/B0076HD2W8?psc=1

Controller

http://www.amazon.com/gp/product/B003ZSP0WW?psc=1

Here's the Product and Vendor ID for the receiver:

Product ID:	0x0291
Vendor ID:	0x045e  (Microsoft Corporation)
Version:	1.14

I don't know how to give that info for the controller as I don't have a way to physically connect the controller to my computer.  

Please forgive me if I'm over-stepping my bounds by even asking, or am making some basic error or asking in the wrong forum.  

-Jon

FranticRain, 2016-01-31:
Okay, so first and foremost, does the wireless adapter light up when you connect it to the computer? Can you attempt to sync the controller to the wireless adapter using the sync buttons on both devices? Can you test the combination on Windows to make sure that you didn't get a broken device?

inasimplerhyme, 2016-02-01:
Both of them lit up.  I hit the button on the controller to search for the receiver, but it didn't work.  But I hadn't pressed the button on the receiver yet.

That worked.  

I feel dumb and ashamed.  But also incredibly grateful to the community here for making this happen.  Thank you so much!!!

FranticRain, 2016-02-01:
Glad it works for you.

closed 2016-02-01 by FranticRain