360Controller/360Controller #221: El Capitan 10.11.3 not supported ?

pyed, 2016-02-01:
I couldn't get my controller to work, I get `No devices found` even tho it is recognized by the system

```
  Product ID:	0x028f
  Vendor ID:	0x045e  (Microsoft Corporation)
  Version:	2.04
  Speed:	Up to 12 Mb/sec
  Location ID:	0x14200000 / 17
  Current Available (mA):	1000
  Current Required (mA):	100
  Extra Operating Current (mA):	0
```

I tried both Beta 5 and 3, both not working.

pyed, 2016-02-01:
I tried Xbox one controller and it worked fine with Beta 5

FranticRain, 2016-02-01:
That is a new controller, surprisingly. 0x028e is the normal Xbox 360 controller, but this one is one higher for some reason. Must have made a minor revision or something? This should fix your problem. Let me know if it doesn't work. https://drive.google.com/file/d/0B5PLCKoMd4zGaVBzYnkzMlVxNGs/view?usp=sharing

pyed, 2016-02-01:
sadly it didn't

[image]

thanks for the effort.

FranticRain, 2016-02-01:
Wait a minute. 0x028f is the PID of the Wireless controller. Are you plugging it in to the computer via a Play and Charge cable? That doesn't work. Xbox One controllers will convert to wired when connected, but a wireless 360 controller will ALWAYS be wireless. No wired connection will change that.

pyed, 2016-02-01:
sorry about that 😓 thank you for your time.

closed 2016-02-01 by pyed