360Controller/360Controller #222: No devices found

ExplosiveTortuga, 2016-02-05:
MadCatz Call of Duty GamePad
Product ID: 0xf025
Vendor ID: 0x1bad

I am running El Capitan 10.11.3 and the latest release 0.15 (beta 5).  I was able to get this controller to work when I was running Yosemite but not since upgrading.  When i plug it in my computer recognizes it but the driver still gives me "No devices found."  Is there any fix for this?

FranticRain, 2016-02-05:
Does this solve your issue? https://drive.google.com/file/d/0B5PLCKoMd4zGTURCdXNJYlU1MmM/view?usp=sharing

ExplosiveTortuga, 2016-02-05:
THANK YOU! Works great!

closed 2016-02-05 by ExplosiveTortuga

simbrams, 2024-01-05:
Hey @Drewbadour I have the same issue with the same controller but your link is dead:
Product ID:	0xf026
Vendor ID:	0x1bad

I'm on Version 0.16.11 

Could you provide the fix again please ?