360Controller/360Controller #223: When will the El Capitan be supported?

polonsil, 2016-02-07:
Hi, kindly advise approximate date for Mac Os El Capitan to be supported? Thanks.

FranticRain, 2016-02-07:
El Capitan has been supported since its first official release.

closed 2016-02-07 by FranticRain

polonsil, 2016-02-07:
Thanks. But there is no way I can hook my Xbox 360 gamepad with my Mac... After installing driver, restarting OS I plug in the gamepad, turn it on and it is just not recognized. I tried like 5-6 latest driver versions. Is there anything I can do to make it work?

FranticRain, 2016-02-07:
Can you get your controller's vendor and product ID from: Apple Button -> About This Mac -> System Report -> USB, find your controller. Just to make sure, this is a wired controller. Wireless Xbox 360 controllers connected with a play and charge cable will not work.

polonsil, 2016-02-07:
Thanks for being supportive, please check: 
Product ID :	0x028f
Vendor ID :	0x045e  (Microsoft Corporation)

FranticRain, 2016-02-07:
That's a wireless 360 controller. It will only work with a wireless adapter. The play and charge kit doesn't turn it into a wired controller, it is only a wireless controller. Xbox One controllers do change to wired controller when connected, but not Xbox 360 controllers.

polonsil, 2016-02-07:
Oh, I see now. Again, thanks so much!