360Controller/360Controller #224: Unable to get Controller to work with Minecraft

OriginalDAN, 2016-02-08:
Wasn't sure if I should have added a response to previously raised issue or create new post - I've done both, sorry if this is wrong...

I just got a mac, and wanted to play minecraft with controller. I see the controller working in the controller test section, however when I go into minecraft control settings it doesn't respond to any of the control pad inputs. Is there anything you can do to help me? I tried having pretend to be a 360 ticked and unticked. Thanks

FranticRain, 2016-02-08:
The PC version of Minecraft does not support controllers. It only supports mouse and keyboard inputs. You could remap your controller inputs to keyboard and mouse inputs using something like Enjoyable or Joystick Mapper.

closed 2016-02-08 by FranticRain

GB-CG, 2019-11-04:
I may add that Enjoyable 1.2 has analog issues with Minecraft 1.14, while Joystick Mapper does not, therefore Enjoyable really needs an update.