360Controller/360Controller #226: Logitech f310 xinput mode compatibility?

killercroc357, 2016-02-10:
wondering if this works with the logitech f310 gamepad, it uses xinput like a 360 controller?

FranticRain, 2016-02-10:
Can you get your controller's vendor and product ID from: Apple Button -> About This Mac -> System Report -> USB, find your controller. This controller should already be supported.

Current Entries are:
PID: 0xC21D
VID: 0x046D

killercroc357, 2016-02-10:
Yup those are the same pid/vid, but on a game called "the Wolf among us" its direct input mode its buttons are jumbled and mess up, but I found on some forums people were able to play it fine with a 360 controller and this driver, figured maybe it'll work without me having to buy a 360 controller since the logitech has a xinput mode

killercroc357, 2016-02-10:
also thanks for answering so fast

FranticRain, 2016-02-10:
If you test the controller using the preference pane, Apple Button -> System Preferences -> Xbox 360... does the controller work as it should? Because if it works correctly in the preference pane, its a game-level issue. You can look for remapping tools inside the game or try using the remapping features built into the driver. Beyond that, we can't do anything.

kevineaves, 2016-06-01:
@FranticRain it works correctly in the preference pane.

FranticRain, 2016-06-01:
@kevineaves If it works in the preference pane, and not in the game, its an issue with the game, not with a driver. There isn't anything we can do about it.

FranticRain, 2016-08-09:
Closing out unfixable game specific issues.

closed 2016-08-09 by FranticRain