360Controller/360Controller #227: Wireless Receiver doesn't pair with controller on El Capitan

magnusr, 2016-02-11:
I have an Xbox 360 Wireless Controller for Windows that I'm trying to get to work on my Mac. The controller does not seem to pair with the receiver - the lights on the Guide button just spin for a while and then switch to blinking when I press the link buttons on the receiver and the controller. The receiver and the controller have never been linked before.
The receiver does show up in the Bus Probe tab of the USB Prober ([USB Bus Probe.txt](https://github.com/360Controller/360Controller/files/125856/USB.Bus.Probe.txt)), and it gets detected by the 360Daemon (I made a build with debug prints), but I don't find any kext related to 360Controller in the Kernel Extensions tab of the USB Prober ([Kernel Extensions.txt](https://github.com/360Controller/360Controller/files/125857/Kernel.Extensions.txt)).

FranticRain, 2016-02-11:
I have to ask, does it work on Windows? You may just have a bad receiver.

kudit, 2016-02-11:
I was having the exact same problem.  Solution was to plug in to a different USB port on the computer.  I don't know if switching back would cause the issue again (I haaven't been brave enough to try that risk trying that)!

magnusr, 2016-02-11:
User error. Replace user and press any key to continue.
Also: don't just press the link button on the receiver - hold it until the light starts blinking.

magnusr, 2016-02-11:
(but big thanks to both of you for being so quick to help out!)

closed 2016-02-11 by magnusr