360Controller/360Controller #228: Not detecting Afterglow Gamepad with El Capitain 10.11.3

WizGnome, 2016-02-11:
I installed 0.15 beta 5 and connected my controller but it was no detected.

Looking in Info.plist idVendor and idProduct were different to that shown in System Information - Product ID: 0xfafd Vendor ID: 0x24c6

I edited info.plist to change idProduct to 64253 and idVendor to 9414 but it is still not detected - did I miss something or is there another issue?

FranticRain, 2016-02-11:
You have to recompile and re-sign before changes to Info.plist take effect. This version of the driver should work for you: https://drive.google.com/file/d/0B5PLCKoMd4zGY19KN2NzN0pvUGc/view?usp=sharing

closed 2016-02-11 by FranticRain

WizGnome, 2016-02-11:
Thanks, but it did not work - the product and vendor id's were swapped - product id is 64253 and vendor id 9414 but in your info.plist they are swapped  :(

FranticRain, 2016-02-11:
Whoops. I do that ever once in a while. here you go. https://drive.google.com/file/d/0B5PLCKoMd4zGQkExNU11RUNXRkk/view?usp=sharing

WizGnome, 2016-02-11:
That fixed it - working now :)