360Controller/360Controller #229: Wired XBox360 Controller Problems with El Capitan 10.11.3

cooperwelch, 2016-02-12:
Hey guys, thanks for the work you've done on this driver so far - really appreciate that this project is still alive and being updated. Unfortunately I'm encountering issues with the latest version of El Capitan, 10.11.3. I've installed the latest beta release, 0.15 (beta 5), and I'm using a wired XBox360 Controller. I plug the controller in and 1 of 2 things happens - either I get an endlessly blinking center ring of lights, or the lights will blink for a couple seconds and then the controller will shut off, no lights at all. Any thoughts? I'm not nearly technical enough to know where to begin on anything like this. Thanks again.

FranticRain, 2016-02-12:
Can you get your controller's vendor and product ID from: Apple Button -> About This Mac -> System Report -> USB, find your controller. I'll need that before I can help you further.

cooperwelch, 2016-02-12:
Sure thing...looking under the USB section of the System Report window, there is a parent entry called BRCM20702 Hub. The info for that is as follows:

Product ID: 0x4500
Vendor ID:  0x0a5c  (Broadcom Corp.)

Underneath that is a child listing called Bluetooth USB Host Controller:

Product ID: 0x8289
Vendor ID: 0x05ac  (Apple Inc.)

If neither of those are what I need to look for, then I guess the controller isn't showing up on my list of USB devices. The only other thing I see is an entry for the Apple Internal Keyboard / Trackpad. (Controller is plugged in and doing its blinking thing btw)

FranticRain, 2016-02-12:
That is not a good sign. OS X doesn't even see your controller. If OS X can't see it, we can't do anything with it. You better check if it works with a Windows machine, because I think that controller is busted.

cooperwelch, 2016-02-12:
Ah ok. I was wondering if that might be the case. I'll definitely check it out. Thanks for the quick replies!

closed 2016-02-12 by cooperwelch