360Controller/360Controller #231: Hori Arcade Pro V Xbox One Controller support (2016)

DEM3NTED, 2016-02-17:
Hey guys when I connect my brand new arcade stick with the latest driver one button presses all buttons
It shows up as wired Xbox one controller 
Even when I move my joy stick it also presses buttons A,B start and select buttons when I remap 
It glitches to multiple buttons that I didn't press
Any help with source file adjustments?

FranticRain, 2016-02-17:
Can you get your controller's vendor and product ID from: Apple Button -> About This Mac -> System Report -> USB, find your controller. I'll need that before I can help you further.

DEM3NTED, 2016-02-17:
Vendor: ID: 0x0f0d (HORI CO., LTD.)
Product ID: 0x0063

Thanks so much!!!

FranticRain, 2016-02-17:
This device has been supported to several iterations now. Did you use any other Xbox One drivers before using this one? They can interfere with this driver. Which ones did you use?

DEM3NTED, 2016-02-17:
Previous install of 360 controller The one before the beta releases then I installed xone osx didn't work uninstalled it, basically uninstalled everyone to the latest it has the same problem arcade stick is near new and works perfect with Xbox one btw

FranticRain, 2016-02-17:
Ah ha, how did you "uninstall" Xone-OSX? Did you use the `uninstall.command`?

DEM3NTED, 2016-02-17:
Yes I did its not there in system preferences after I did that

DEM3NTED, 2016-02-17:
I'm on El Capitan

FranticRain, 2016-02-17:
So there is no `Xone Driver.kext` in `/Library/Extensions`. The main problem here is that there is another user that has reported the Hori Real Arcade Pro Hyabusa as usable, and it has the exact same VID/PID. So it is very likely that the issue is on your system.

DEM3NTED, 2016-02-17:
No I have no Xonr Driver.kext in there just checked

DEM3NTED, 2016-02-17:
Xone*

FranticRain, 2016-02-17:
Do you have any normal Xbox One controllers you can test the driver with? I guess 360 controllers might also work, but I would really rather prefer you tested it with a standard Xbox One controller.

DEM3NTED, 2016-02-17:
Yes I just tested my Xbox one wireless controller it works so smoothly with that but the hori stick alone triggers the dpad and both anolog sticks if I move right it randomizes a direction usually triggering start and select back and forth while the rb is being held at all timed the A B RT LT LB RB Y & X famicom buttons trigger the dpad in random direction everytime it's pressed while also holding done on left and right analog stick also left and right analog sticks always have a highlighted purple box

DEM3NTED, 2016-02-17:
Yeah all the buttons are just randomizing every key

DEM3NTED, 2016-02-17:
Apologize for the convoluted description

DEM3NTED, 2016-02-17:
I*

FranticRain, 2016-02-17:
Okay, so my best guess is that the fight stick returns an irregularly sized packet. This packet is not the same as a normal Xbox One packet or Elite packet. Do you have a Windows PC you can do some testing with? I'm going to need you to sniff the controller's packets, using the tools Wireshark and USBPcap.

DEM3NTED, 2016-02-17:
Yes I do will download now

DEM3NTED, 2016-02-17:
But I have the unofficial driver installed on windows is that ok?

DEM3NTED, 2016-02-17:
From Major Nelson

FranticRain, 2016-02-17:
That's perfectly fine. Follow the instructions on this guide (http://www.definedbehavior.com/reverse-engineering-the-xbox-one-controller/) until you get a capture out. When capturing, just press a couple of the face buttons in an order you can tell me. For example, just tap `A B X Y` in that order and stop capturing. Then post that capture here.

DEM3NTED, 2016-02-17:
X-Y-(RB)-(LB)
A-B-[RT]-[LT]-

0000: 1b 00 00 0e 23 85 ff ff ff ff 00 00 00 09 00                     ....#... .........
0010: 01 03 00 01 00 81 01 08 00 00 00 03 20 02 04 80   ......... .... 
0020:01 00 00  ...

FranticRain, 2016-02-17:
I don't know where you got that data, but it is completely unusable. Start with the controller plugged in, find what port it is on with USBPcap, select that port, name the file, press a couple of buttons, stop the cap, and then give me that file. Don't just paste it raw text.

FranticRain, 2016-02-17:
I made a slightly better guide on how to sniff packets. Let me know if you have any problems with it. https://franticrain.github.io/sniffing/

DEM3NTED, 2016-02-18:
https://www.mediafire.com/?yxm3emt033b76v0

FranticRain, 2016-02-18:
Try this version of the driver out: https://drive.google.com/file/d/0B5PLCKoMd4zGS3dFTmVmODJSTWs/view?usp=sharing

DEM3NTED, 2016-02-18:
the remapping feature lets me add in the b command before exiting out before i could enter Y and X but it has the same issue as before randomizing a bunch keys while one key is pressed.. anything I could do from here? id even donate for your time.. thanks

DEM3NTED, 2016-02-18:
heres a video of how it reacts http://www.mediafire.com/watch/11713y77syzq55c/Video.MOV

FranticRain, 2016-02-18:
Please stop messing with the bindings until you notice a difference in the behavior of the driver. That feature is precarious enough as is, and you throwing bad inputs at it definitely isn't going to help the issue. For all I know, I fixed it with that version and you've jacked up the rebinding tool so it still appears messed up.

Now, could you please get me a packet sniff that is JUST the A, B, X, Y buttons, in that order. If you start pressing random buttons and moving sticks around, I can't tell what I'm looking at. This is an odd sized packet, I just have to figure out why it is odd sized, and why the driver doesn't like it. Knowing exactly where A, B, X, and Y are will really help.

DEM3NTED, 2016-02-18:
ok will do shortly, keep in mind i didn't bind 1st if you can see from the video the analog alone triggers both ways the analog even triggers the start and select buttons it'll press left and right on  the dpad while i just hold left or right and rb usually is always held down just throwing that out there in all versions

FranticRain, 2016-02-18:
Trust me, you don't know how this works. That is a valid symptom of something going REALLY wrong with rebinding.

DEM3NTED, 2016-02-18:
pressed A.B,X,Y in that order 2x decimals wouldnt show up unless i did https://www.mediafire.com/?ahhjxpzwb9i5phh

FranticRain, 2016-02-18:
Okay, try this version of the driver. This one adds some debug stuff and removes some features that might be causing issues. You can find the debug output from the driver in your Console.app, which is located in `Applications/Utilities`. When you open up Console.app, click `system.log` in the left hand sidebar. Everything pertaining to what we want should be prefixed: `Hori - `

https://drive.google.com/file/d/0B5PLCKoMd4zGanlTeURtMjV5VXc/view?usp=sharing

DEM3NTED, 2016-02-18:
Works great just not as fast Responding as the Xbox one controller THANKS SO MUCH !! - love

DEM3NTED, 2016-02-18:
Rt and Lt can't be used btw

FranticRain, 2016-02-18:
Yeah, parts of it not working was kind of the point. I stripped a bunch of stuff out. Now we are going to slowly re-add parts so we can figure out what part in particular it is what was causing issues. (Spoilers: Its almost assuredly rebinding.)

DEM3NTED, 2016-02-18:
ok just let me know bro

FranticRain, 2016-02-18:
Triggers and Southpaw but no Rebind: https://drive.google.com/file/d/0B5PLCKoMd4zGbUNTTlQ0a1VmS2s/view?usp=sharing
Rebind but no triggers and no southpaw: https://drive.google.com/file/d/0B5PLCKoMd4zGM1ZReTBRVHNvTzg/view?usp=sharing

Let me know how these work. When using the one that supports rebinding, make sure to press the "Reset Rebinding" button.

DEM3NTED, 2016-02-18:
triggers do not work on the first one basically both work the same i can also rebind on the one you said i can't it works just as it did on the 1st working build you made

DEM3NTED, 2016-02-18:
need me to capture RT and Lt buttons?

FranticRain, 2016-02-18:
You can rebind things in the preference pane, but they don't actually do anything unless you are using the one specifically marked as supporting rebinding.

Can you rebind buttons to different values using the rebinding on the supported driver? I assume when you say that you mean that it doesn't work with a traditional controller, or does the fight stick have triggers on it? The code that impacts triggers shouldn't have changed for normal controllers whatsoever. Can you test it using something like Steam Big Picture to make sure?

DEM3NTED, 2016-02-18:
the fight stick has triggers on it but i guess you didn't activate the response i just tested it with opener its working ok

FranticRain, 2016-02-18:
The triggers should probably work in both builds. I'm going to need you to tell me if the triggers work with a controller on any of these 3 builds (you can choose.) I also need to know if the triggers on the fight stick work. Just simple yes/no please.

DEM3NTED, 2016-02-18:
I'm using "Rebind but no triggers and no southpaw" build
but the triggers are working it vibrates when pressed on my xbox one controller
checked the "Triggers and Southpaw but no Rebind" build
but the trigger buttons on my arcade stick are not working

FranticRain, 2016-02-18:
Okay yeah, I'll need a sniff of just the triggers on the fight stick. Just press both of them a couple of times, I should be able to figure it out just fine.

DEM3NTED, 2016-02-18:
RT,LT in that order pressed multiple times https://www.mediafire.com/?lmnc6j282jd5p85

FranticRain, 2016-02-18:
The triggers are not reporting any data. Do the triggers work on windows in games and such?

FranticRain, 2016-02-18:
OH. They are buttons! Oh now I get it. I've never used a fight stick before, sorry. This explains so much, let me try something out real quick.

FranticRain, 2016-02-18:
Okay, this should work. Let me know. https://drive.google.com/file/d/0B5PLCKoMd4zGYXo5TkF5b0NCMGs/view?usp=sharing

DEM3NTED, 2016-02-18:
fucking AWESOEME!!! it even responds way faster man

DEM3NTED, 2016-02-18:
perfect

FranticRain, 2016-02-18:
Okay, great. I'll clean this code up tomorrow and get you a final version.

DEM3NTED, 2016-02-18:
Thanks a ton

FranticRain, 2016-02-18:
This version should speed up driver performance a little bit and is cleaned up quite a bit from what I was doing earlier. If you're looking to test latency between the controller and a program, don't use the preference pane. It's pretty bulky and not built for speed.

https://drive.google.com/file/d/0B5PLCKoMd4zGb2xxWUJnT1pib0k/view?usp=sharing

Going to go ahead and close this issue unless you have a problem with that build.

closed 2016-02-18 by FranticRain