360Controller/360Controller #233: WineSkin+ePSXe right trigger constantly pressed

stevenvachon, 2016-02-21:
OS 10.11.3
WineSkin 2.6, Wine 1.9.2
ePSXe 2.0

Using a wireless 360 controller with play-and-charge cable in USB.

The right trigger is constantly pressed in the gamepad config.

FranticRain, 2016-02-21:
Wireless 360 controllers are always wireless. They do not work through a play and charge kit. How in the world is that working? Also, why are you using wine to use ePSXe when you could be using [Mednafen](http://mednafen.fobby.net/) on OS X, or even better yet Mednafen via [OpenEmu](http://openemu.org/)?

I'm going to tag this as "game specific" because, as usual, we can't fix individual programs. That's on the devs of those programs. Not to mention we extra can't fix anything on Wine.

MaddTheSane, 2016-02-21:
This is mainly how Wine implements Xbox 360/One controllers is different than what Windows applications expects.

There is a Wine plug-in that maps SDL2 controllers to XInput. The repository is [here](https://github.com/KoKuToru/koku-xinput-wine).

stevenvachon, 2016-02-21:
@FranticRain mednafen does not improve graphics and I'm not playing on a CRT.

@MaddTheSane It's happening with PCSXR too (Mac version). I don't think it has to do with Wine or ePSXe.

FranticRain, 2016-02-21:
So does the controller exhibit this right trigger issue in the preference pane? Or something like big picture? Because you still completely avoided my question about how in the world you are using a wireless 360 controller wired via a play and charge kit. That doesn't work.

stevenvachon, 2016-02-21:
I updated my original comment. I'm just avoiding the use of batteries.

Right trigger does not exhibit this behaviour in preferences pane.

FranticRain, 2016-02-21:
If its not in the preference pane, its almost assuredly a game level issue. All I can really tell you is that OpenEmu will work with your controller, and I highly recommend it. If you insist on still using Wine, all I can recommend it using that plugin that @MaddTheSane recommended, and checkout out configuration windows for controller settings.

stevenvachon, 2016-02-23:
I'm no C/C++ guy. I don't know how to use `koku-xinput-wine`.

FranticRain, 2016-08-09:
Closing out unfixable game specific issues.

closed 2016-08-09 by FranticRain