360Controller/360Controller #234: Controller not recognized.

zero334, 2016-02-21:
Hi, first of all great work!
I bought this wired controller, and it doesn't seem to be supported.

Vendor name: Trust
Product name: Predator

Product ID: 0x0003
Vendor ID: 0x0810
Version: 1.06

FranticRain, 2016-02-21:
This driver is for controllers that work with Xbox consoles. I don't see any Xbox controllers from Trust. Are you sure this is an Xbox 360 or Xbox One controller?

zero334, 2016-02-21:
Think so, you can switch between DirectInput and XInput with a switch.
It is an older controller not listed on their site.

FranticRain, 2016-02-21:
Okay, give this a shot then: https://drive.google.com/file/d/0B5PLCKoMd4zGbHFwb25kQ0JkQ1U/view?usp=sharing

zero334, 2016-02-21:
Works great :dancer:

FranticRain, 2016-02-21:
Awesome.

closed 2016-02-21 by FranticRain