360Controller/360Controller #237: Help with OS X 10.8.3

daisyfllower, 2016-02-21:
Hi! I just got a play-and-charge Xbox One controller to use with my Steam games (I had gotten sick of my Playstation controller), hoping to use some sort of platform to make it work. I have OS X 10.8.3 and it seems that the newest platform doesn't work with anything less than OS X 10.9. HELP ME. I don't want to return this controller! Oh and I know nothing of kernel drivers and kext so sorry this is going to be difficult.
D

TyanColte, 2016-02-26:
@daisyfllower does your play and charge plug into the bottom of the controller (I.E. proprietary xbox one connector) or the top of the controller (I.E. standard Micro USB?)

FranticRain, 2016-02-26:
@TyanColte It has nothing to do with the controller. Pre-OS X 10.9, kexts were stored in /System/Library/Extensions, and were signed differently. Now they are stored in /Library/Extensions with the different signing certificates. I have no idea how old kexts were treated, and I started developing kexts in the days of 10.9, so one of the other contributors would have to help him, but its mostly gone ignored because more people update their operating systems, especially when it is free.

There are other issues regarding this kind of thing. #216, #42, #34. But its just not a major issue since we all updated our systems. You might be able to build it yourself and put it in /System/Library/Extensions, but I honestly don't know for sure.