360Controller/360Controller #241: Build failed: Code Sign error

ascrane, 2016-02-28:
I just cloned the repo and attempted to build, getting the following error:

```
=== BUILD TARGET 360Daemon OF PROJECT 360 Driver WITH CONFIGURATION Release ===

Check dependencies
Code Sign error: No code signing identities found: No valid signing identities (i.e. certificate and private key pair) were found.

** BUILD FAILED **

The following build commands failed:
	Check dependencies
(1 failure)
******** BUILD FAILED ********
```

I'm on OS X 10.10.4 running Xcode 7.0.1 – I was under the impression I could build and run an unsigned version of the project.

FranticRain, 2016-02-28:
You are correct in that assumption, but the build script tells the driver to build using a signing signature. If you were to build with Xcode instead of the build script, it would have compiled just fine. In order to fix this, you need to do several things. First, you need to change every target in the .xcodeproj to have a "Code Signing Identity" of "Don't Code Sign" on Release. Next, you need to open the Install360Controller.pkgproj with Packages.app, and tell it not to sign in the "Project" menu. After that, you need to disable code signing, which is explained in the Readme, so I won't repeat that.

So as you can see, you can build an unsigned version, though I fail to see why you would want to do it with the build script, because thats how we do signed release builds. Therefore, complaining about that not immediately supporting no code signing certificate seems more than a bit dense.

closed 2016-02-28 by FranticRain

ascrane, 2016-02-28:
Thank you for the response, but this was definitely one of the rudest responses I've ever received from a developer/maintainer.

Steps to reproduce:
# Open the README
# Navigate to the Building section
# Interested in buildling the standard artifact, a .dmg, not a .pkg. The README suggests to run the build script and that's about it. Unless someone has built signed packages with Xcode in the past, I don't think your suggestions are obvious.

I'm sure there are other, external factors, related to this project or your personal life that might lead you to create a response like that. But taking that as the default attitude towards someone that is interested in your project is ill-spirited at best.

FranticRain, 2016-02-29:
You know what, you're right, I was a jerk. I apologize.

No one ever bothers to touch that README. Its tremendously out of date, and nearly impossible to understand. I've been meaning to rewrite it for a long time, but have never gotten around to it.

So here's exactly what you need to do to build the driver without a signing certificate. A process I DO NOT RECOMMEND ANYONE UNDERTAKE but you seem quite emphatic about the idea.

This project consists of 9 products and 2 build types.
**Products:**
* Whole Driver
* DriverTool (Helps with things)
* 360Daemon (Does some background stuff)
* Feedback360 (Force feedback plugin)
* Pref360Control (The preference pane)
* 360Controller (The driver)
* Wireless360Controller (The wireless controller part)
* WirelessGamingReceiver (The wireless receiver part)

**Build Types:**
* Debug
* Release

All of this information can be found by quickly opening the .xcodeproj, and checking some pretty basic stuff in Xcode. You can even search the properties of each target for the name "sign" to find out information about the signing certificates required. Now, as one might suspect of software that gets released, it is signed with a certificate. This is the kind of thing that would happen in a "Release" build. While on the other hand, a "Debug" build gives you all the same stuff a "Release" build does, but without the whole signing part weighing you down.

So you have a couple of options from this point:
1. Remove all the signing certificate requirements from the release build. I already told you how to do that one.
2. Make a debug build
  1. Open up Xcode and build the driver using the "Whole Driver" target. Easy!
  2. Adjust the build script to do a debug build instead. Completely undocumented!

Now depending on what you just did, you've got one of two things. Either:
A. All of the products built out into a folder.
or
B. A pkg, that installs all of the products into their proper spots, and reboots the computer.

If you have A. here's what you need to do:
* Put `360Controller.kext`, `Wireless360Controller.kext`, and `WirelessGamingReciever.kext' into `/Library/Extensions`.
* Make sure to `sudo chown -R root:wheel ` all of those after you've put them in.
* Put `Pref360Control.prefPane` into `Library/PreferencePanes`
* Make a symbolic link from `/Library/Extensions/360Controller.kext` to `/System/Library/Extensions/360Controller.kext`
No .pkg required, but also no .dmg.

If you have B. here's what you need to do:
* Double click that thing, install the driver, reboot.
The .pkg is required, but no .dmg.

If you have A and want to make a .dmg, google's probably your best bet. If you have B and want to make a .dmg, what are you doing? You already have the installer. The only reason you'd want to put it into a .dmg is to distribute an unsigned driver. I really hope that's not what I just helped you do. But if you insist, run the `makedmg.sh` in the `Install360Controller` directory. This will make a .dmg that contains the .pkg.

Now, you may notice that the driver still doesn't work. That's likely because you haven't disabled codesigning. That's actually covered in the README because I recently updated that part. So follow those instructions, and then you can have the exact same thing you could have gotten out of [stuff on the release page](https://github.com/360Controller/360Controller/releases), but with more security risks!

Now that I've taken the time to write all this out, I'd appreciate it if you'd tell me why on earth you insist on building the driver yourself.

ascrane, 2016-02-29:
Drew, thank you for the in-depth response. Sorry if this was a hassle, but I felt it was worth making an issue about since it's the default path for someone looking to build this project themselves without paying for a developer license and setting up code signing in Xcode.

Now that I understand the project structure, I realize the .pkg is in fact what I wanted to build all along. After looking at the release .dmg, it simply contains the .pkg. This makes sense, considering that it's basically just unloading a handful of kexts and a preference pane on the install target. I was interested in building the project myself simply so I could look over the source and verify its origin before installing it.