360Controller/360Controller #242: Afterglow Support for Xbox One Controller

Luxiary, 2016-02-29:
Title pretty much says it all. I have an Xbox One controller that came with the console, I tried it with this driver and it worked flawlessly. I recently bought an Afterglow controller for my xbox one as well and I wanted to give it a go with this but it's not working. It doesn't even detect my controller. Is there some sort of solution to this?

FranticRain, 2016-02-29:
Can you get your controller's vendor and product ID from: Apple Button -> About This Mac -> System Report -> USB, find your controller. I'll need that before I can help you further.

Luxiary, 2016-02-29:
The Product ID is 0x0139 & the Vendor ID 0x0e6f.

FranticRain, 2016-02-29:
Hmm. This one should already be working. Give this a shot and then let me know if it works: https://drive.google.com/file/d/0B5PLCKoMd4zGbjF0dHpLQUljT1E/view?usp=sharing

Luxiary, 2016-03-01:
It's working! You're a lifesaver. Thank you!

FranticRain, 2016-03-01:
Awesome.

closed 2016-03-01 by FranticRain