360Controller/360Controller #245: Battery Level

metatrevor, 2016-03-27:
Is there a way to get the wireless joystick battery level as a percentage (not bars) from the preferences pane ?

TyanColte, 2016-04-04:
I'm sure it'd be possible, but I don't know if that's something the devs would implement. Maybe as a configurable option. What do you think @FranticRain @RodrigoCard ? Seems promising.

FranticRain, 2016-04-04:
Its theoretically possible. It looks like the controller reports 0-255 as the battery level. Thats all minor changes to the preference pane though, and I'm not the one you want to talk to about preference pane stuff.

tomlube, 2016-04-10:
Hey FranticRain, not to be a dick but I saw you got a Wireless adapter just before the break recently. How is progress using the controller wirelessly? :)

FranticRain, 2016-04-10:
@tomlube We have a whole thread for that: https://github.com/360Controller/360Controller/issues/156#issuecomment-207828002

FranticRain, 2016-08-10:
Got a little prototype of this I finally put together. What do you guys think?

[image]

RodrigoCard, 2016-08-14:
looks good :)

FranticRain, 2016-08-14:
Just pushed this feature. Will be in the next release.

closed 2016-08-14 by FranticRain