360Controller/360Controller #249: Xbox One Controller not Working on OS X 10.11.4

wizcreations, 2016-04-03:
I've been using this driver (release 0.15 beta 5) for an Xbox 360 controller without issues.  I attempted to use an Xbox One controller but iMac does not even recognize the controller.  Is there something I need to do to enable support for the Xbox One controller?

FranticRain, 2016-04-03:
Can you get your controller's vendor and product ID from: Apple Button -> About This Mac -> System Report -> USB?

wizcreations, 2016-04-03:
My computer does not recognize the XB One controller.  It does not appear in the list of USB hardware.

The 360 controller does appear (although I'd guess that is not what you're asking for).  Product ID: 0x028e.  Vendor ID: 0x045e  (Microsoft Corporation)

FranticRain, 2016-04-03:
Find a different USB cable and connect your Xbox One controller with that. It looks like you are using a micro USB charge cable, not a data cable.

wizcreations, 2016-04-03:
Oh man, I'm dumb!  Thanks for reminding me of that.  Works perfectly when I grab the right cable.

FranticRain, 2016-04-03:
No problem.

closed 2016-04-03 by FranticRain