360Controller/360Controller #250: Controller Inputs Do Not Prevent Screen Saver

buffaloseven, 2016-04-04:
When using this driver combined with a wireless receiver and a X360 pad, the screensaver will activate in some programs.  While an application can be set to disable the screensaver, not all developers (I'm looking at you, Unity devs) set the flag correctly.  Is there a way where inputs from a controller fed through this driver would register in a way that interrupts the screen saver timer, similar to how the mouse and keyboard does?

4ndv, 2016-04-04:
I have the same issue when playing Bastion

fedxc, 2016-04-04:
I have the same issue, on the other hand I saw a patch note from a game I play and they fixed this on their end. So it might be a game related issue.

buffaloseven, 2016-04-04:
@fedxc Yes; developers can disable the screen saver from an application. It would just be nice if it was possible (not sure if it is) if the inputs from the controller (save for analog sticks; perhaps exceeding threshold values for them) registered in a similar way to mouse/keyboard inputs and prevented the screen saver.

fedxc, 2016-04-04:
@buffaloseven understood! Hope there is a solution for this :)

slime73, 2016-05-23:
As mentioned, OS X (and other operating systems like Windows as well) explicitly don't do this automatically. It's not the driver's responsibility to do this, and doing so would mean this driver behaves differently from every other joystick/gamepad driver in OS X (including Apple's own built in ones).

Application developers are the ones responsible and you should contact them instead.

FranticRain, 2016-08-09:
Closing out unfixable game specific issues.

closed 2016-08-09 by FranticRain