360Controller/360Controller #251: No Way To Donate To Code Signer

josephquigley, 2016-04-07:
Whoever is paying Apple for the privilege of using their keys to sign the .kext needs a donation button. ;)

emmanueljarri, 2019-06-23:
How can we make donations to Drew?

FranticRain, 2019-06-27:
I know this is like 3 years later, but I'm leaving the project soon due to personal issues with how the driver is being used by third parties. If you'd like to donate to me via PayPal as thanks for all of the work I've done over the past 5-ish years, you're more than welcome to. The stuff I learned along the way was well worth it, but I certainly won't stop you from donating. [The PayPal link is here](https://www.paypal.com/cgi-bin/webscr?cmd=_donations&business=MVEQJJ9S2EVTS&currency_code=USD&source=url)