360Controller/360Controller #252: Issue with Yosemite 10.10.5 & Afterglow pad

yamsellem, 2016-04-09:
Release 0.11, 0.12, 0.13, 0.14 & 0.15 (beta 5) have fail on Yosemite 10.10.5 with an Afterglow pad.
Following [this video](https://www.youtube.com/watch?v=K2bRKEO9b6k) solved the issue.

In a nutshell:
* [Tattiebogle 0.08](http://files.tattiebogle.net/360/360ControllerInstall.0.08.dmg) Xbox 360 controller drivers
* [KextBeast](http://www.tonymacx86.com/downloads.php?do=file&id=310) to install the edited .kext file
* [UnTattie script](http://www.mediafire.com/download/35xl6qjahi32qwg/UnTattieboglev2.sh) to uninstall the drivers

d235j, 2016-04-09:
@yamsellem please try uninstalling the driver and installing this build:
https://github.com/360Controller/360Controller/issues/242#issuecomment-190340151

@FranticRain could you promote a new beta?

FranticRain, 2016-04-09:
Should be up. Hopefully I didn't miss label changes.

DashyFTW, 2016-04-15:
My "Afterglow Gamepad for Xbox 360" does not work with the latest beta, or any version for that matter. My Product ID is "0x0301" and my Vendor ID is "0x12ab". What occurs is that my controller simply is not recognised by the software whilst it works 100% perfectly running Windows 7, 8.1 and 10. It is, however, recognised in the "System Report...", under USB in El Capitan.

FranticRain, 2016-04-15:
@DashyFTW [Here you go.](https://drive.google.com/file/d/0B5PLCKoMd4zGLTBhaVN2Z19UZXc/view?usp=sharing)

DashyFTW, 2016-04-16:
I thank you @FranticRain. Very much appreciated. +1

FranticRain, 2016-05-15:
Closing due to inactivity. Controller should be added.

closed 2016-05-15 by FranticRain