360Controller/360Controller #253: No devices found

DrStrangeLuv, 2016-04-12:
Hello, I purchased a generic Xbox 360 controller from Amazon, hoping that it would work fine as a basic controller on Yosemite 10.10.5, but maybe not..?  Trying for platform games like Limbo.  I've been trying a bunch of the drivers - latest being 0.15 from post that sounded like a similar issue ([#222](https://github.com/360Controller/360Controller/issues/222)), but I still can't get the controller to be recognized.  The Xbox button lights up for a second when plugging in the usb, but then nothing.  Any ideas?  Thanks so much.

FranticRain, 2016-04-12:
Can you get your controller's vendor and product ID from: Apple Button -> About This Mac -> System Report -> USB?

DrStrangeLuv, 2016-04-12:
Thanks.  I'm seeing a couple entries that may be what you're asking for, but maybe not..?  Sorry in advance if not.

USB Hi-Speed Bus -> Hub:
Product ID: 0x0024
Vendor ID: 0x8087 (Intel Corp)

USB Hi-Speed Bus -> Hub:
Product ID: 0x2513
Vendor ID: 0x0424 (SMSC)

Any of this help?  I can screen grab if that'd help?

FranticRain, 2016-04-12:
Does the device have a wire coming out of it natively? Or are you using some extra adapter to make it wired?

DrStrangeLuv, 2016-04-12:
Yes, it's natively wired.

FranticRain, 2016-04-12:
Right now it looks like the controller just don't work. If OS X isn't seeing it in that window I told you about, we can't do anything about it. Check and make sure it works with a 360 or a Windows PC.

DrStrangeLuv, 2016-04-12:
Ok, I must not have gotten the right one.  Bummer.  But thanks for the help!  I'll try again.

closed 2016-04-12 by DrStrangeLuv