360Controller/360Controller #254: Problem witch a game

chicochaleur29, 2016-04-16:
Hello
I have a problem witch my xbox one controler for play "The Walking Dead Michone" for mac
When i start the game, the start bouton don't work, it's the LS Click work.
In game, when i push the LS control on the left, at the screen, i go on the right....
I have this problem only this game
I have the last upadate of your software (0.15 beta 6) and OSx 10.11.5.
Can you help me?
Thanks

FranticRain, 2016-04-16:
In your "Xbox 360 Controller" Preference Pane, go to the "Advanced" tab and check the "Pretend to be an Xbox 360 Controller" box.

chicochaleur29, 2016-04-16:
Yes, i have test to, but don't work... :(

FranticRain, 2016-04-16:
Then we can't do anything for you. This is a problem with the game, not with the driver. Sorry.

FranticRain, 2016-08-09:
Closing out unfixable game specific issues.

closed 2016-08-09 by FranticRain