360Controller/360Controller #256: Help/Support with Marvel Heroes 2016 (steam/Mac)

anoandetc, 2016-04-19:
Hi, first of all, I appreciate your work. I was directed to your driver, and despite the problem with MH, it's still the easiest to use. If you can solve this issue, there will be a lot of happy mac users thanking you (there's no conclusive solve to this problem that I can find, and MH's staff isn't planning to help). 

The game doesn't appear to register either of the trigger buttons, which is crucial for powers in game (R2 or L2 + X, Y, or B = Power 3, 4, etc). 

 Also, for whatever reason, I always have to invert my left stick's Y-axis to get the movement right (a minor issue, comparatively). 

 The driver's preferences window does show that the buttons are working, however. I've tried re-mapping. I'm just not really sure what I'm doing, to be honest.   

I've tried multiple wired Xbox 360 controllers (off-brand and on) with no luck.  

10.11.4 El Capitan. 

As this is my first post, I'm not sure what other information to include. I hope you can help! Thank you, in any case.

FranticRain, 2016-04-20:
Unfortunately, we can't do much here. A driver basically gives games a framework that they can hook in to. We can only change things at a driver level. It is impossible to build in game-specific fixes for things like this. The best I can recommend is using something like [Joystick Mapper](http://joystickmapper.com/) to map specific buttons and triggers to keyboard buttons. That should allow you to use all of your abilities, but still maintain the full directional movement of the joysticks. I've never done this myself, but it would work.

anoandetc, 2016-04-23:
Appreciate the help!  I'll give joystick mapper a go. Been using the controller for Don't Starve lately, and it works great.

SirSteven2005, 2016-05-30:
Is there any FREE applications?

FranticRain, 2016-05-30:
If you are looking for free options, you may want to look at [Enjoyable](https://yukkurigames.com/enjoyable/) and [Enjoy2](https://github.com/fyhuang/enjoy2). I cannot speak for their quality level, or if they will even work at all, though. Use at your own risk.

FranticRain, 2016-08-09:
Closing out unfixable game specific issues.

closed 2016-08-09 by FranticRain