360Controller/360Controller #257: Latest xbox one controller firmware

DJAlterego, 2016-04-20:
So i find myself seeking advice after hours of searching for a answer... I have 2 controllers (xbox one) that i use and both were updated to latest firmware and now neither work. When i plug them in the controller only vibrates once, and thats it. nothing sees it, light does not come on at all. hold the xbox logo and it flashes as if it was looking for a console to turn on and then stops. Does anyone have a workaround for this?

FranticRain, 2016-04-20:
Well, if your controllers worked previously and now they don't, you're stuck until I figure out the new protocol. If you haven't used them before, I'm going to need their vendor ID and product ID.

So if they worked before, I'm going to need a sniff [following the procedures I have outlined here.](https://franticrain.github.io/sniffing/index.html)

DJAlterego, 2016-04-20:
FranticRain Thanks so much for a quick reply! Yes, indeed it was working great till one of my kids did me a favor and updated them both lol. When i get a min, i will hop into bootcamp and get the requested info.

FranticRain, 2016-04-20:
Oh yeah, specifically in this case, plug your controller in to find it listed in USBPcap, but before you start sniffing, unplug the controller. After the sniff has started, plug it back in. That will give me the data I need for the controller init packets.

Moltress2, 2016-04-26:
Im having the same problems as DJAlterego.

FranticRain, 2016-04-26:
The please get sniffs, just like I told him to.

pogzie, 2016-05-01:
Im not sure if using the controller on XBoxOne or Windows 10 would mandate you to update the firmware of the controller, but I was wondering if we could add a warning on the main page not to update the firmware.

FranticRain, 2016-05-01:
I would hope that most people would know that using 3rd party hacks means that firmware updates might break them. Where would you propose I put it? Possibly in a "Warnings" section above the "About" section in the README? I can tell you for a fact that almost no one reads those things. I've put several such style warnings in READMEs that have been completely ignored. I don't mean to disparage you. I just know people like to click through to downloads without reading anything. It's human nature.

Ideally, this would only be a short term issue if anyone would just give me a sniff. Its probably only 30 minutes of work, but I need the data from a controller that suffers from this issue. I used my controller with an Xbox One for several hours today, and I cannot recreate the issue. Granted, I used the controller wirelessly.

FranticRain, 2016-05-01:
In fact, I just forcibly updated a 2013 controller through a fully updated Xbox One and I cannot recreate the issue. I cannot solve this without a third party sniff.

Moltress2, 2016-05-01:
I will try to do a sniff. You said you have guidelines to do it right?

FranticRain, 2016-05-01:
[This is the process, right here.](https://franticrain.github.io/sniffing/index.html) Make sure you start sniffing before plugging in the controller. The init packets on first connect are the most important part of this.

Moltress2, 2016-05-01:
https://drive.google.com/file/d/0BwF7_jJCVKiOaU8teERMMlJ4UjA/view?usp=sharing

FranticRain, 2016-05-01:
@Moltress2 Did you press anything while the controller was connected? It looks like there is 1 original packet and a bunch of identical packets. I need to see some buttons presses or I can't see where the button information is located in the packet.

Moltress2, 2016-05-01:
I did push the buttons, but ill try agian.

Moltress2, 2016-05-01:
When I press the controller buttons, is there supposed to be something that shows up in the terminal window?

FranticRain, 2016-05-01:
No, you will see nothing from USBPcap. It will just show up in the sniff if you do it correctly.

Moltress2, 2016-05-01:
https://drive.google.com/folderview?id=0BwF7_jJCVKiOeUFvVlhiT1lDU0U&usp=sharing

Aight, attempt number 2

FranticRain, 2016-05-01:
Okay, that one has some usable data in it. I'll take a look.

Moltress2, 2016-05-01:
Ill try my best to work with you and help further.

FranticRain, 2016-05-01:
Can you get one sniff where the controller starts plugged in, you tap A, B, X, Y and the stop the sniff?

Moltress2, 2016-05-01:
how do I stop the snif? do I just do the control+c?

FranticRain, 2016-05-01:
Correct.

Moltress2, 2016-05-01:
aight, refresh the drive page. it should show up now

FranticRain, 2016-05-01:
Was the controller plugged into the same port the entire time? In this third sniff, its from the source: 1.11.1 In the second sniff, there are no packets matching the pattern from sniff 3, but there is a LOT of noise from 1.5.1. Sniff 1 contains data from neither 1.11.1 nor 1.5.1. Do you have multiple USB hubs on your computer, because I need a sniff from a much quieter USB hub. You'll notice in my tutorial, there is a "1" with a BUNCH of stuff on it, but the controller is on "2." That makes this IMMENSELY easier. I'll need one that specifically sniffs the init packets. So connect the controller, find the controller, disconnect the controller, start the sniff, leave it for a few seconds, end the sniff. Because from that 3rd sniff it looks like the button packets are identical. This is fantastic news.

Moltress2, 2016-05-01:
yes, it is plugged in the same one the entire time. Ill take out the other USB's I had in. I only have a 1, not 2 USBPcaps. Check the drive again.

FranticRain, 2016-05-01:
There are still no init packet in this sniff. There are several data sends from 1.12.1, but never a data send TO 1.12.1. After you start the sniff, give it a little time and then plug the controller in. It must be that this sniff is starting slower than you are plugging the controller in.

Moltress2, 2016-05-01:
So, name the document. plug in the controller, press the buttons. once the buttons are pressed. Then hit control+c?

Moltress2, 2016-05-01:
Sorry, once I get this we will get this done quick. I dont mean to waste your time. I have never done this before XD

FranticRain, 2016-05-01:
Name the document, start the sniff, wait a couple seconds, connect the controller, press A, wait for a couple seconds, disconnect the controller.

Its perfectly fine. For all I know, you're doing it perfectly and there is some black magic that doesn't require an init packet. Basically, I need the A press just to make sure what device the controller is in the sniff, but other than that, button presses don't matter.

Moltress2, 2016-05-01:
does the sniff start automatically, or do i do something after naming it and pressing enter?

FranticRain, 2016-05-01:
It starts after you name it and press enter.

Moltress2, 2016-05-01:
Aight, dropped into drive

FranticRain, 2016-05-01:
There we go! That one has it! Let me put something together real quick.

Moltress2, 2016-05-01:
Aight, sounds good. Ill stick with that combo in the future. Also, it may also be helpful to update your "how to sniff for xbox one packets page" because that may be helpful to other users also.

FranticRain, 2016-05-01:
Try out this version and see if it works: https://drive.google.com/file/d/0B5PLCKoMd4zGc3lySUhiaE04S3c/view?usp=sharing

Moltress2, 2016-05-01:
aight, give me a moment

FranticRain, 2016-05-01:
I'm not entirely sure that's a bug. The left had side of the controller has a larger vibration motor than the right hand side. These motors must be unbalanced, or they will counteract each other's movement. To make absolutely certain, go into the preference pane and look at the circles that fill up when you pull the triggers. Make sure that the right trigger is filling up the same amount as the left trigger does when one of them is fully pressed. Don't press both of them at the same time, because that might cause issues that will ruin the test, for weird, backend, packet reasons.

If they both fill up the same amount, I'd be pretty confident in saying that its operating as intended, but I'll leave this open until you figure it out.

**To Everyone Who Experienced This Issue:**
If you could please fill out a survey of sorts so I can figure out why I can't reproduce this, that would be great.

1. What brand is your controller? (Ex. Microsoft, PDP, Afterglow, etc).
2. Does your controller have a 3.5mm headphone jack?
3. How did you update your controller? (Ex. Windows 10 or Xbox One)
4. Are you in the Windows Insider program or is your Xbox in the preview program?

Moltress2, 2016-05-01:
1. microsoft controller 
2. No
3. Updated thru xbox one
4. I am part of the preview program.

pogzie, 2016-05-01:
@FranticRain I agree, im just a little bit concerned that your attention might veer away from the wireless xbox one adapter support by working on stuff like this because people keep updating their firmware.. :)

FranticRain, 2016-05-01:
@Moltress2 That's exactly the situation I'm in, and my controller worked perfectly fine. Weird.

@pogzie Wired controllers always come first. If MS is updating the firmware, its probably for a good reason, so its best to always make sure that wired controllers work. Wireless controllers are a luxury item in this business, so I'd much rather work on the wired ones.

Moltress2, 2016-05-02:
I played Left 4 dead with my xbox one controller. It seems to work perfectly! Thank you frantic!

D1no, 2016-05-18:
I just want to add, that the Microsoft Xbox One Elite controller, that used to work before, is not working anymore after the new controller firmware update through the xbox one. Also the alternative driver mentioned above (https://github.com/360Controller/360Controller/issues/257#issuecomment-216016098) is not changing anything. Plugging in the controller only lets it vibrate once and thats it. I have a spare xbox 360 controller - but would be great if there is a way to get it working again.
[image]

FranticRain, 2016-05-18:
@D1no Going to need a sniff out of you. [This is the process.](https://franticrain.github.io/sniffing/index.html)

D1no, 2016-05-19:
Damn, I don't have a windows machine on my hand and I just dumped my win10 image from my macbook because of disk space needs.... great timing. If there's another way, let me know. I am abroad right now

FranticRain, 2016-05-19:
Unfortunately, it has to be on Windows. That is currently the only way.

FranticRain, 2016-07-07:
I need someone who needed these new init packets to try out this build and see if it still works properly for you: https://drive.google.com/file/d/0B5PLCKoMd4zGSnNRVUV5V3AwTVk/view?usp=sharing

TTimo, 2016-07-07:
This doesn't seem to work. My 360 controllers still work fine, but the XBox One is in the same state as initial report - doesn't initialize.

I ran a capture on Windows 10 here:
https://www.dropbox.com/s/kzvx70sdzb701t0/xbone.pcap?dl=0

FranticRain, 2016-07-07:
That sniff appears to be on a pretty busy USB hub. Unfortunately, its hard to get anything useful out of it. But knowing that these init packets don't work on the "new firmware" is really useful, albeit disappointing. It would be *extremely* helpful if my controllers worked one way or the other, because for some dumb reason they work with both styles init packets, ugh. I'll keep looking into this.

TTimo, 2016-07-07:
It's the front port on my desktop. I'll try to capture something less busy.
Note that this controller never worked on OSX for me. I probably got it
with the latest firmware out of the box.

TTimo

FranticRain, 2016-07-07:
If it didn't work before for you, does that include 0.16.1? It has the init packets that *should* work with the "new firmware" controllers.

TTimo, 2016-07-07:
Try this one: https://www.dropbox.com/s/56ho0ocdi7x6jke/xbone2.pcap?dl=0

I never tried 0.16.1 - I was running 0.16 until this latest drop. Should I try 0.16.1? Is it different from the one you uploaded earlier?

FranticRain, 2016-07-07:
Yes. There are currently three different init packets rolling around. 0.16 has style A, 0.16.1 has style B, and the Google Drive link has style C.

That sniff is still no good. When you're selecting a number with USBPcap, take note that everything under the same number is still on the same "hub." If you move around the controller from the front IO to the back IO, it is likely it'll still be on the same usb hub.

TTimo, 2016-07-07:
Are you sure you are looking at the right file? After I unplugged the other stuff, I'm pretty sure it's alone on that hub:
https://gyazo.com/bac92fe33fecd9e13c16467dc5729039

I did another capture just in case: https://www.dropbox.com/s/8o78rogskbjjaeg/xbox3.pcap?dl=0

I also verified that 0.16.1 doesn't work.

FranticRain, 2016-07-07:
Oh its because you are using a wireless adapter! Wireless adapters currently aren't supported. We are just talking about wired controllers here. If you are looking for information about using the wireless adapter, check out #156

TTimo, 2016-07-07:
Derp. Ok .. once wired the controller works fine, with 0.16.1 and with the latest dmg.

FranticRain, 2016-07-07:
That's excellent news. I'll need confirmation from someone who previously had issues with 0.16 before I can call this a success though. There are too many variations on how people's controllers react to these init packets. Such a hassle.

RodrigoCard, 2016-07-07:
I just got a 2015 controller and updated the firmware.
I will test 0.16 and the new version

RodrigoCard, 2016-07-09:
Just tested my new 2015 controller (forza edition, [this one](http://compass.xbox.com/assets/26/6b/266b5688-9507-4c27-a3e1-e79e29fcc89d.jpg?n=xbox-one-forza-motorsport-6-wireless-controller_hardware_image_960x540_01.jpg)) with updated firmware and noticed a behavior change:

**0.16   :**
- plug the controller and cable, works as expected.

**0.16.1:**
- plug the controller and cable, vibrates, doesnt work.
- second try, same behavior.
- decided to press the Home/Guide button, it works!
- unplugged/ re plugged the cable, works.
- as long as the xbox button is blinking, it connects and works normally.

FranticRain, 2016-07-09:
Marking this as resolved with the release of 0.16.2.

closed 2016-07-09 by FranticRain