360Controller/360Controller #258: Different Product ID

dalanmiller, 2016-04-22:
Just wanted to check you were in the know about this different Product ID for a Wireless Xbox One Controller. 

This was working for me within the past two weeks with Hyper Light Drifter, but now does not work. 

```
Controller:

  Product ID:	0x02dd
  Vendor ID:	0x045e  (Microsoft Corporation)
  Version:	2.03
  Serial Number:	7EED8DD07671
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Microsoft
  Location ID:	0x14100000 / 13
  Current Available (mA):	1000
  Current Required (mA):	500
  Extra Operating Current (mA):	0
```

FranticRain, 2016-04-22:
This isn't a different product ID. This is the product ID of the 2015 controller (the one with the 3.5mm jack). I'm assuming you recently used this controller with an Xbox One or Windows 10 PC and updated your firmware. This is the same issue as #257 Please get a sniff as described in that issue.

FranticRain, 2016-05-02:
This problem has been resolved in #257 and will be pushed to a release soon. Please check that out to solve your problem.

closed 2016-05-02 by FranticRain