360Controller/360Controller #260: Issue with AfterGlow AP.2 on Os x 10.10.5

Boulmagik, 2016-04-23:
Hello, i'm trying to make my controller work with several versions of the drivers, and it don't work. I can see the product id : 0x1414 . I have tried to replace the id product of a controller in your drivers (info.plist) and apply an editor of .kekt . But my controller is still not detected. Of course, i have tried to restart my computer several times. Sorry about my bad english it is not my mother tongue. Thank you

FranticRain, 2016-04-23:
I need your Vendor ID and your Product ID and then I can fix this issue.

Boulmagik, 2016-04-23:
Product ID : 0x1414 
  Vendor ID : 0x0e6f (if i have translated correctly, i think it is this you talked about)

FranticRain, 2016-04-23:
Try this: https://drive.google.com/file/d/0B5PLCKoMd4zGVFRQLWFCNkk1ZGM/view?usp=sharing

Boulmagik, 2016-04-23:
The product is detected, thank you, but the wireless connection don't work, or is lose instantly in 2 seconds. Do i forget to mention that this device is wireless or is that an issue that depends of the builder of this controller ?

FranticRain, 2016-04-23:
You must use the controller wired. Wireless does not work at this time.

closed 2016-04-23 by FranticRain