360Controller/360Controller #267: Xbox Controller S (X08-69873) "Unknown device release 121"

jamesathey, 2016-05-06:
With version 0.16 on OS X 10.11.4, this is what I get when I plug in my Xbox Controller S:

system.log:

kernel[0]: Unknown device release 121
kernel[0]: start - unable to find the interface

In Hardware->USB in System Information, the device appears as:

Hub:

  Product ID:	0x0288
  Vendor ID:	0x045e  (Microsoft Corporation)
  Version:	1.21
  Speed:	Up to 12 Mb/sec
  Location ID:	0x14200000 / 9
  Current Available (mA):	1000
  Current Required (mA):	100
  Extra Operating Current (mA):	0

This appears to be the same problem @tmecklem had, and which @RodrigoCard asked him to create an issue for. @kasbert, any ideas?

kasbert, 2016-05-06:
Would you be able to get USB endpoint identifiers?
You can get them using linux or maybe some usb prober application.
6.5.2016 5.59 ip.

FranticRain, 2016-05-06:
Actually, just try out this build real quick: https://drive.google.com/file/d/0B5PLCKoMd4zGMzJiV29haHN4bk0/view?usp=sharing

authorfunction, 2016-07-20:
I can confirm that I have the same problem with a modded original Xbox controller. I posted a new issue before seeing this: https://github.com/360Controller/360Controller/issues/319

edrex, 2016-09-20:
I am seeing this in the log as well when plugging in an original xbox ("S") controller:

```
9/19/16 5:19:15.000 PM kernel[0]: Unknown device release 100
9/19/16 5:19:15.000 PM kernel[0]: start - unable to find chatpad interface
9/19/16 5:19:15.000 PM kernel[0]: handleReport 1 
9/19/16 5:19:15.000 PM kernel[0]: 01 03 06 
```

The driver seems to work normally however.

FranticRain, 2019-01-15:
This issue is being closed due to inactivity. We can't fix issues if we don't have enough information. So periodically we close down issues that can't be resolved if they seem to have been abandoned by their creators. If your issue still persists, please open a new issue after thoroughly consulting the README and providing all information as required by the README.

closed 2019-01-15 by FranticRain