360Controller/360Controller #268: How to use driver?

ubernex, 2016-05-07:
I installed the program and restarted my mac, plugged in my controller, but nothing is happening. I am incredibly stupid. Help please!

FranticRain, 2016-05-07:
What type of controller are you using? Is it a wireless Xbox 360 controller? If you are plugging it in using a "Play and Charge Kit" cable, that only provides power. It does not turn a wireless controller into a wired one. You will still need a wireless adapter to make it work.

If it isn't that, I'm going to need your controller's product and vendor ID. You can find those in: Apple Menu -> About This Mac -> Overview Tab -> System Report... -> Hardware Section -> USB. There should be a device in there listed as "Controller". Click on it, and then vendor and product IDs should be listed there.

ubernex, 2016-05-07:
I was trying an xbox one controller – does the driver work with it?

FranticRain, 2016-05-07:
It does. There are a couple things you have to be careful of though. Make sure you are using a "data" USB cable. Some USB cables are only wired to charge devices. This will allow the controller to turn on, but it will not send data to the computer. If your controller is listed in that menu I mentioned, then it is connected with a "data" cable. Secondly, is this a third party controller? If so, I'm definitely going to need that vendor and product ID. Third, have you recently updated the firmware? You'll have to use a pre-release build to fix certain issues with the latest firmware on the Xbox One controller.

ubernex, 2016-05-07:
The controller is listed in the menu, it's a controller straight from the console box with product ID 0x02dd and vendor ID 0x045e. I think I used the release firmware, which version do you recommend?

Also thanks for helping!

FranticRain, 2016-05-07:
Try this version of the driver and let me know if it fixes the problem: https://drive.google.com/file/d/0B5PLCKoMd4zGc3lySUhiaE04S3c/view?usp=sharing

ubernex, 2016-05-07:
What should I do once it's installed and my computer is restarted?

FranticRain, 2016-05-07:
All you have to do is plug in the controller. That will start the controller up and then it should work in games. You can check for sure by checking the preference pane in your System Preferences.

ubernex, 2016-05-07:
Oh awesome thank you!

FranticRain, 2016-05-07:
Make sure to let me know if it does or doesn't work so I can close this issue.

ubernex, 2016-05-07:
It's responding the the system preference page but not on CS:GO or Super Meat Boy

FranticRain, 2016-05-07:
Go into the preference pane, click on the "Advanced" tab, and enable "Pretend to be an Xbox 360 controller." That should fix your problem.

ubernex, 2016-05-07:
Still not functioning :(

FranticRain, 2016-05-07:
Glad to hear it!

closed 2016-05-07 by FranticRain