360Controller/360Controller #269: No device found - Nacon Controller

Szedria, 2016-05-09:
Hi, I'm using Nacon GC-100XF and I tried to search the support, but it doesn't seems it works yet. It should be working as Xbox 360 Controller as it can switch between XInput and DirectInput8. 

  Product ID:	0x55f0
  Vendor ID:	0x11c9  (Monster Cable Products, Inc.)
  Version:	1.00

I use driver version 0.16.
Thank you in advance.

FranticRain, 2016-05-09:
Let me know if this fixes your problem: https://drive.google.com/file/d/0B5PLCKoMd4zGTi1KNk1GVm1UdnM/view?usp=sharing

Szedria, 2016-05-09:
No, it doesn't working yet. I tried to fresh-install driver and used different USB port also. The driver hasn't recognized it yet.

There are some changes though. For what I observe:

**Before:**
- The LED only partially lit (only on top-left quarter of circle).
- On OpenEmu: I could switch using left stick or D-pad by pressing Mode button.
- On Steam Big Picture Mode: I could set mapping of the gamepad and it works on some games.

**After:**
- The LED partially lit half circle _ONLY IF_ I rebooted my Mac while plugging-in the gamepad. It shows left half of circle is lit. When I press the Mode button, the right half of circle is lit.
Its indicates DirectInput mode is active and able to switch between analog and digital mode (according to user manual).
- On OpenEmu: OpenEmu fails to open.
- On Steam Big Picture Mode: The configs of active button 1-4, i.e on Xbox Controller A-B-X-Y, so I had to remap it again and same games work.

**IF** I press the D/X Button, the After-state changes into Before-state and nothing changes when I repress D/X Button again, e.g I can play on OpenEmu again or have to remap on Steam Big Picture Mode. 

I guess it might good to add about games on Steam I tried:
- _Playable, Works well:_ Danganronpa: Trigger Happy Havoc
(although the configurations changes between analog and digital mode, but not affect the gameplay)
- _Playable, Works, but have to do remapping in-game, tad slow:_ Rats, Time is Running Out!
- _Unplayable, the configs are totally messed up from Steam Big Picture Mode:_ Mark of the Ninja & Cook, Serve, Delicious!

**Remark:**
I tried to give infos as much as I can. Please let me know if additional infos needed. Thank you again in advance!

FranticRain, 2016-05-09:
This driver should light only the top left quarter of the circle to indicate that you are using controller number 1. But what you are telling me is that the controller worked with 0.16 in the "before" state, but after I gave you this new version of the driver, it behaves in the "after" state. Nothing I did should have changed how the lights illuminate or anything like that.

It seems like there is something else in play here. The driver should ONLY work with the controller in XInput mode. DirectInput should either not work, or give you undefined controller inputs. Furthermore, the driver should never pair with any device that isn't listed in the Info.plist. Your controller was not listed in the Info.plist and should never have illuminated in the first place.

We've had trouble with these DirectInput/XInput devices before, for example the F310 from Logitech, and I have no idea how to fix them. They SHOULD be complying to the XInput standard, which this driver is looking for, but it sure seems like they aren't. So I'm not at all sure how to go about fixing this problem.

Szedria, 2016-05-10:
Maybe coz it's "plug-and-play" gamepad, hence built-in driver? Even without 360Controller driver, I could play on OpenEmu.

Forgot to mention that I tried it on Virtual Box (Windows 10) and it behaves similar to "After-state" (just able to test it with Mark of the Ninja though). It would go disconnected when I press D/X Button. So basically it might doesn't work well in XInput even if I try on native Windows PC.

I guess my best bet is just grab Xbox Controller.
_Out of topic question:_ What should I consider when choosing to buy Xbox Controller? e.g Do I need to buy Microsoft Xbox 360 Controller _for Windows_ or the one for Xbox also work?

FranticRain, 2016-05-10:
That's not quite what Plug and Play is. Plug and Play means that it just complies USB standards so that it will match with the correct driver on your computer. The reason it works with OpenEmu is because it complies with DirectInput, which is a standard OS X natively supports.

If you want to get an Xbox 360 controller, and Xbox 360 controller will work. Just remember that wireless controllers will require a wireless adapter and wired controllers just plug in. There is no way to make a wireless controller wired or vice versa.

Szedria, 2016-05-10:
Uhm I see. Thank you for your efforts. I appreciate it.

FranticRain, 2016-08-09:
Closing this, since this is an unsupported type of controller.

closed 2016-08-09 by FranticRain