360Controller/360Controller #272: Cannot uninstall on Mac OS X

kas, 2016-05-11:
`System Preferences -> Xbox 360 Controllers -> Advanced Tab -> Uninstall button.`

I have the version from [tattiebogle](http://tattiebogle.net/index.php/ProjectRoot/Xbox360Controller/OsxDriver)
I've followed those steps and there is no Advanced Tab in the Xbox 360 Controllers section.  What am I missing?

FranticRain, 2016-05-11:
That is not our driver. We do not support other drivers.

closed 2016-05-11 by FranticRain

kas, 2016-05-11:
Just realized this.  Thanks

kas, 2016-05-11:
You may want to change the project description to reflect that this is not TattieBogle's driver!

FranticRain, 2016-05-11:
It is based on that driver, it has the same core and identifier. But if you look at the release page, you see that our version number is higher, and released more recently. They are effectively identical, but this one is currently maintained. We support the current version of the current driver, not old an outdated ones for years ago.

Additionally, this information IS in the README. It is the second paragraph. As well, nothing about that website links to this Github page. In fact, it links to a DIFFERENT Github page. Assuming that they are the same thing is an impressive stretch of logic.

kas, 2016-05-11:
Right on.  Habitually I will read the repo description before looking at the README and I'm sure you understand and do the same.  The description begins saying that this is TattieBogle's driver.

Any visitor who is looking to get their XBox controller working on their mac is going to go to TattieBogle's webpage and download that.  Many will assume that this is the repo for that download.  There is not much of a stretch of logic in assuming that these two projects are the same.

FranticRain, 2016-05-11:
But your argument is that people who don't read the README wouldn't know that they were different so we should add it to the README?