360Controller/360Controller #275: Controller Does'nt work on OSX el capitan

geppolo, 2016-05-15:
Hello I've installed the software on an Imac retina 5k 27'' with OSX el capitan 10.11.4 , when I connect the controller nothing happens. I can feel just a vibration but the light stays off.
Anyone has a solution? thanks.

FranticRain, 2016-05-15:
Sounds like you are using an Xbox One controller. It sounds like its also likely that you recently updated your firmware on that controller. Try installing the version of the driver from here: https://github.com/360Controller/360Controller/issues/257#issuecomment-216016098

jaredkent, 2016-05-27:
It's the USB cable not the program, controller, or firmware. I ran into this same issue and after spending over an hour on tech support with windows (I ran into the issue on windows 10 in bootcamp, so with native controller support) we determined that a different cable solved the problem. It appears as though not all USB cables are equal. I had a few that would just cause the controller to vibrate but not turn on. I'm sure you have a few around the house as most of us do at this point. Just try swapping a few out and find one that works.

FranticRain, 2016-05-27:
That is because you used why is referred to as a "charge" cable. They only provide power and do not allow for data transfer. You have to find a "data" cable in order to actually transfer data from the controller to the computer. The best way to check this in the future is to go to the Apple Menu -> About This Mac -> Overview Tab -> System Report... -> Hardware Section -> USB. If the controller isn't listed there, then you are using a charge cable with the controller. If it is listed, then the vendor and product ID could be transferred because it is a data cable.

closed 2016-05-27 by FranticRain