360Controller/360Controller #280: 0.16 Xbox360 completely hosed..

larianstudios, 2016-05-16:
Ok simple tests with 0.16 and DOS:EE

1. Xbox one controller is connected to 10.10.5
2. Game sees Xbox One controller 1118/721 and Rumble works 
3. (BUG) Rumble options are Default/None
4. Game works correctly
======
1. I click on Advanced/Pretend
2. Game sees Xbox 360 wired 1118/654 and Rumble is not working (BUG)
3. (BUG) Rumble is not working in SysPref panel. Rumble options are Default/None/Trig/Both (Wrong?)
4. Game works correctly but rumble is not working

== I disconnect controller/connect it back/Close syspref

1. (BUG) 360 pretend checkbox is not saved

FranticRain, 2016-05-16:
I've been messing with some of that stuff in prepartation of a 0.16.1. Try this version out and tell me if you see the same issues. https://drive.google.com/file/d/0B5PLCKoMd4zGLWtjb25Nd2lXejA/view?usp=sharing

FranticRain, 2016-05-16:
Using that version of the driver, here is what I see.
Triggers only cause rumble on the "Controller Test" tab. That is on purpose. Not a bug.
All four rumble options are available for both "Pretend 360" and not. Not a bug.
Rumble options are maintained through disconnects and closes of the preference pane. Not a bug.
Pretend 360 is maintained through disconnects and closes of the preference pane. Not a bug.
After toggling Pretend 360, rumble stops working on the Controller Test tab. However, given enough time and presses of triggers, it will eventually resume. This could be considered a bug, but we are also hot swapping the core functionality of a driver here. It's not a clean process. I'll see if there is a way to fix it, but it just seems like plugins really don't like the hot swap.

larianstudios, 2016-05-16:
Ouch! This new one is rely wicked. I see controller 'flash' for a second then it disappears then appears again and this goes forever.

FranticRain, 2016-05-16:
That is interesting. Everyone I've given this version to before has not complained about that. There are at least 10 people using 0.16.1 pre-release at this point.

larianstudios, 2016-05-16:
System.log - unknown device release 101

larianstudios, 2016-05-16:
Some other stuff related to Sys Pref:

ay 16 18:36:30 Platons-MBP System Preferences[406]: -[CUIThemeRendition newImageFromCSIDataSlice:ofBitmap: usingColorspace:] Unsupported pixel format in CSI got 2147449848
May 16 18:36:30 Platons-MBP System Preferences[406]: Unable to create unsliced image from csi bitmap data.
May 16 18:36:30 Platons-MBP System Preferences[406]: -[CUIThemeRendition newImageFromCSIDataSlice:ofBitmap: usingColorspace:] Unsupported pixel format in CSI got 2147449848
May 16 18:36:30 Platons-MBP System Preferences[406]: Unable to create unsliced image from csi bitmap data.
May 16 18:36:30 Platons-MBP System Preferences[406]: -[CUIThemeRendition newImageFromCSIDataSlice:ofBitmap: usingColorspace:] Unsupported pixel format in CSI got 2147449848
May 16 18:36:30 Platons-MBP System Preferences[406]: Unable to create unsliced image from csi bitmap data.
May 16 18:36:30 Platons-MBP System Preferences[406]: -[CUIThemeRendition newImageFromCSIDataSlice:ofBitmap: usingColorspace:] Unsupported pixel format in CSI got 2147449848
May 16 18:36:30 Platons-MBP System Preferences[406]: Unable to create unsliced image from csi bitmap data.
May 16 18:36:30 Platons-MBP System Preferences[406]: -[CUIThemeRendition newImageFromCSIDataSlice:ofBitmap: usingColorspace:] Unsupported pixel format in CSI got 2147449848
May 16 18:36:30 Platons-MBP System Preferences[406]: Unable to create unsliced image from csi bitmap data.
May 16 18:36:30 Platons-MBP System Preferences[406]: -[CUIThemeRendition newImageFromCSIDataSlice:ofBitmap: usingColorspace:] Unsupported pixel format in CSI got 2147449848
May 16 18:36:30 Platons-MBP System Preferences[406]: Unable to create unsliced image from csi bitmap data.
May 16 18:36:30 Platons-MBP System Preferences[406]: -[CUIThemeRendition newImageFromCSIDataSlice:ofBitmap: usingColorspace:] Unsupported pixel format in CSI got 2147449848
May 16 18:36:30 Platons-MBP System Preferences[406]: Unable to create unsliced image from csi bitmap data.
May 16 18:36:30 Platons-MBP System Preferences[406]: -[CUIThemeRendition newImageFromCSIDataSlice:ofBitmap: usingColorspace:] Unsupported pixel format in CSI got 2147449848
May 16 18:36:30 Platons-MBP System Preferences[406]: Unable to create unsliced image from csi bitmap data.
May 16 18:36:30 Platons-MBP System Preferences[406]: -[CUIThemeRendition newImageFromCSIDataSlice:ofBitmap: usingColorspace:] Unsupported pixel format in CSI got 2147449848
May 16 18:36:30 Platons-MBP System Preferences[406]: Unable to create unsliced image from csi bitmap data.
May 16 18:36:30 Platons-MBP System Preferences[406]: -[CUIThemeRendition newImageFromCSIDataSlice:ofBitmap: usingColorspace:] Unsupported pixel format in CSI got 2147449848
May 16 18:36:30 Platons-MBP System Preferences[406]: Unable to create unsliced image from csi bitmap data.
May 16 18:36:30 Platons-MBP System Preferences[406]: -[CUIThemeRendition newImageFromCSIDataSlice:ofBitmap: usingColorspace:] Unsupported pixel format in CSI got 2147449848
May 16 18:36:30 Platons-MBP System Preferences[406]: Unable to create unsliced image from csi bitmap data.
May 16 18:36:30 Platons-MBP System Preferences[406]: -[CUIThemeRendition newImageFromCSIDataSlice:ofBitmap: usingColorspace:] Unsupported pixel format in CSI got 2147449848
May 16 18:36:30 Platons-MBP System Preferences[406]: Unable to create unsliced image from csi bitmap data.
May 16 18:36:30 Platons-MBP System Preferences[406]: -[CUIThemeRendition newImageFromCSIDataSlice:ofBitmap: usingColorspace:] Unsupported pixel format in CSI got 2147449848
May 16 18:36:30 Platons-MBP System Preferences[406]: Unable to create unsliced image from csi bitmap data.
May 16 18:36:30 Platons-MBP System Preferences[406]: -[CUIThemeRendition newImageFromCSIDataSlice:ofBitmap: usingColorspace:] Unsupported pixel format in CSI got 2147449848
May 16 18:36:30 Platons-MBP System Preferences[406]: Unable to create unsliced image from csi bitmap data.
May 16 18:36:30 Platons-MBP System Preferences[406]: -[CUIThemeRendition newImageFromCSIDataSlice:ofBitmap: usingColorspace:] Unsupported pixel format in CSI got 2147449848
May 16 18:36:30 Platons-MBP System Preferences[406]: Unable to create unsliced image from csi bitmap data.
May 16 18:36:30 Platons-MBP System Preferences[406]: -[CUIThemeRendition newImageFromCSIDataSlice:ofBitmap: usingColorspace:] Unsupported pixel format in CSI got 2147449848
May 16 18:36:30 Platons-MBP System Preferences[406]: Unable to create unsliced image from csi bitmap data.
May 16 18:36:30 Platons-MBP System Preferences[406]: Failed to connect (bindingInputColumn) outlet from (Pref360ControlPref) to (NSTableColumn): missing setter or instance variable
May 16 18:36:30 Platons-MBP System Preferences[406]: Failed to connect (linearDeadzoneLeft) outlet from (Pref360ControlPref) to (NSButton): missing setter or instance variable
May 16 18:36:30 Platons-MBP System Preferences[406]: Failed to connect (linearDeadzoneRight) outlet from (Pref360ControlPref) to (NSButton): missing setter or instance variable
May 16 18:36:30 Platons-MBP System Preferences[406]: Failed to connect (xoneRumbleOptions) outlet from (Pref360ControlPref) to (NSPopUpButton): missing setter or instance variable
May 16 18:36:30 Platons-MBP System Preferences[406]: Unable to simultaneously satisfy constraints:
	(
	    "<NSLayoutConstraint:0x61800009f9a0 H:[MyWhole360Controller:0x6180001d4280(465)]>",
[... cut ...]
	Set the NSUserDefault NSConstraintBasedLayoutVisualizeMutuallyExclusiveConstraints to YES to have -[NSWindow visualizeConstraints:] automatically called when this happens.  And/or, break on objc_exception_throw to catch this in the debugger.
May 16 18:36:30 --- last message repeated 2 times ---
May 16 18:36:30 Platons-MBP System Preferences[406]: Error: couldn't find ControllerType
May 16 18:36:30 Platons-MBP System Preferences[406]: No settings found
May 16 18:36:30 Platons-MBP System Preferences[406]: Result of kextstat = 0

larianstudios, 2016-05-16:
Reverted back to 0.16. All OK. (Well, no flashes and cycles)

FranticRain, 2016-05-17:
I fixed the delayed rumble issue. You're welcome to try this one out, but it will likely give you the same issues you had with the last one: https://drive.google.com/file/d/0B5PLCKoMd4zGU19kb2ZKUklBMTg/view?usp=sharing

larianstudios, 2016-05-17:
Exactly.

FranticRain, 2016-05-17:
Here are all of the changes, but with the old init packets. https://drive.google.com/file/d/0B5PLCKoMd4zGTlhSNy1McDNSZ2c/view?usp=sharing

larianstudios, 2016-05-17:
Works... A lot of UI bugs still but they all circle around simple issue - no preferences are saved. Ever.
Where is the PLIST? As for flashes - what is your plan of attack on this issue?

larianstudios, 2016-05-17:
CFPreferencesSynchronize(DOM_CONTROLLERS, kCFPreferencesAnyUser, kCFPreferencesCurrentHost);....

Note that you can only save preferences for “Any User” if you have root privileges (or Admin privileges prior to OS X v10.6). 

How on earth can this work then?

FranticRain, 2016-05-17:
It works perfectly for me and many others. I have no idea what your issue is.

larianstudios, 2016-05-17:
Well the idea is - you can't save from System Prefs to /Library/Preferences.. You can see the PLIST file there. You can, but only using privileged processes. I launched System Preferences as root and indeed it saved all the stuff. But as long as I save them using just System Preferences nothing is saved..

buffaloseven, 2016-05-17:
Simple permissions issue, is it not? Shouldn't preferences be saved to ~/Library/Preferences/ for proper per-user settings?

FranticRain, 2016-05-17:
I've never really spent that much time messing with accessing and retrieving settings, but the driver is very specific about what it can and cannot look at within the file system. For all I know it may not have access to anything in "~".

larianstudios, 2016-05-17:
Well the chain of commands are as follows
360 daemon waits for devices and pushes settings to them. 360 daemon is run as root and is absolutely capable of writing plists. System Preferences pane however is not capable of doing anything like this. It remains complete mystery to me how can @FranticRain save settings correctly. What 360 daemon can do however is propagate System Prefs settings back to plist. It's very likely approach and I was kinda expecting to see it in the code. However.. nope.

FranticRain, 2016-05-17:
Well, it works for me and everyone else who uses 0.16. If you know how to fix it, you're more than welcome to open a pull request, because if I can't recreate it, I'm going to have a very hard time fixing it. So you are the perfect person for the job.

larianstudios, 2016-05-17:
Sure, np. Can you do a simple test for me and if it works I'll be puzzled for life ;)
1. Go to /Library/Preferences/. Do an 'ls' for com.mice.driver.Xbox360Controller.devices.plist. Note its size.

2. Open System Preferences. Change something. Close. Check same file as in 1. Does the file size change?

FranticRain, 2016-05-17:
Here's the exact process I followed:
- Before you asked, I had deleted that file. So I restarted my machine to clear any caching.
- Upon boot, file was still not there
- File was not created until I checked "Swap Sticks" on the "Controller Test" tab. At this point it is 611 bytes.
- Repeatedly unchecking and re-checking the toggle resulted in the same file size. 611 bytes.
- Leaving "Swap Sticks" checked, I unplugged the controller and full quit the preference pane.
- I reconnected the controller, and re-opened the preference pane. "Swap Sticks" was checked. Settings still 611 bytes.
- Checked "Invert X" and "Invert Y" for both sticks on "Controller Test" tab. Still 611 bytes.
- Unplugged controller before restarting computer
- After reboot, preferences is still 611 bytes.
- Re-opened preference pane, all settings were saved, and unchecked all "Invert X", "Invert Y" and "Swap Sticks" check boxes on the "Controller Test" tab. Still 611 bytes.
- Checked the "Pretend to be an Xbox 360 Controller" checkbox on the "Advanced" tab. Still 611 bytes.
- Toggled "Swap Sticks" on the "Controller Test" tab. Still 611 bytes.

larianstudios, 2016-05-17:
I AM puzzled for life!

What does plutil -p on this file shows?

FranticRain, 2016-05-17:
```
$ plutil -p com.mice.driver.Xbox360Controller.devices.plist
{
  "7EED8718CA98" => {
    "BindingDown" => 1
    "ControllerType" => 3
    "RelativeRight" => 0
    "BindingA" => 12
    "BindingRB" => 9
    "RelativeLeft" => 0
    "DeadOffLeft" => 0
    "BindingLeft" => 2
    "BindingX" => 14
    "BindingRight" => 3
    "BindingLB" => 8
    "BindingB" => 13
    "SwapSticks" => 1
    "DeadzoneRight" => 0
    "Pretend360" => 1
    "BindingStart" => 4
    "InvertLeftY" => 0
    "InvertLeftX" => 0
    "DeadzoneLeft" => 0
    "DeadOffRight" => 0
    "BindingUp" => 0
    "BindingLSC" => 6
    "BindingRSC" => 7
    "BindingBack" => 5
    "BindingGuide" => 10
    "BindingY" => 15
    "InvertRightY" => 0
    "InvertRightX" => 0
    "RumbleType" => 0
  }
}
```
Note that at this point, "Pretend to be an Xbox 360 Controller" and "Swap Sticks" are the only toggles checked at this point.

larianstudios, 2016-05-17:
Ok. I'll work on this mystery..

larianstudios, 2016-05-17:
Sigh.. it's 10.11 only feature looks like this. CPrefsD saves the data. On 10.10 0 file size.

electric-monk, 2016-05-17:
Can you clarify what you mean? 10.10 isn't working, 10.11 is fine because CPrefsD is saving the preferences on behalf of Preferences.app and so can circumvent the need to be root?

larianstudios, 2016-05-17:
This is very very surprising and strange. I used Dtrace to confirm that in 10.11 CFPrefsd is saving preferences on behalf of Preferences.app. On 10.10 I see similar tries from CFPrefsd which for unknown reason fail.

The outcome is 10.11 - everything works, no idea how it will work on non-admin account (probably won't)
The owner of file in /Library/Preferences is user.

On 10.10 so far nothing works (as I was kinda expecting).

larianstudios, 2016-05-18:
10.9 same story. I am contacting Apple for explanation of this phenomena.

electric-monk, 2016-05-18:
10.9 is like 10.10 or 10.11? ;)

larianstudios, 2016-05-18:
Like 10.10..

larianstudios, 2016-05-24:
Although Apple will explain us what and why happened, the idea still remains the same - settings are not saved pre-10.11. I don't think it will be a cool idea to move settings per-user as it's very non-trivial and not recommended to move driver settings to per-user level.

Instead I am planning a rewrite in the style of https://github.com/MaccaTech/PostgresPrefs/tree/master/PostgreSQL

with a 'Lock' panel and overall GUI bugfixing. The approximate time of code starting ~ end of this week.

FranticRain, 2018-06-01:
Settings bugs should be fixed in the latest release.

closed 2018-06-01 by FranticRain