360Controller/360Controller #281: how to use the driver in my code

piaoxu93, 2016-05-16:
I want to write a program which can receive the input of 360 joystick. I've installed this driver and it works, so, how I can use it in my code?

larianstudios, 2016-05-16:
You have to either use something like SDL or use typical HID IOKit API. Basically what driver does is that it exposes 360 device as a HID compliant device which might stream its data (buttons, sticks) to your app.

FranticRain, 2016-05-16:
Depends on what language you are using, what you are making, and a million other different things. Look for OS X controller libraries in your language of choice. We can't tell you how to program.

piaoxu93, 2016-05-16:
Thanks, I'm using C++, and I found SDL is the right thing for me.

closed 2016-05-16 by piaoxu93