360Controller/360Controller #283: Any plans to let this driver simulate direct input into Xbox 360 input?

cdes, 2016-05-25:
I love using PS3 Controller for games.  I'm using it for both Windows and Mac.

On Windows, there's an app called MotionInJoy that can make my PS3 controller simulates a Xbox controller.

On Mac however, there's nothing that can do that.  Only the impractical apps that let my controller simulates keyboard and mouse.

I wonder if it's technically possible.

Like if you allow the user to bind any OS recognized input (even keyboard) to simulate in the binding panel. 

So, is it technically possible? or is it impossible?

FranticRain, 2016-05-25:
This is not feasible. This is a USB driver. Not a bluetooth, ethernet, etc. protocol driver. Adding support for all directinput controllers would mean multiple drivers for multiple protocols. Developers of games and apps should be supporting directinput in their applications. It is THE controller protocol in OS X, and if they aren't supporting it, they aren't supporting controllers like they should. This is and continues to be a driver that lets Xbox series controllers work with OS X, not one that fixes compatibility issues because developers can't program worth a crap.

Additionally, the upkeep required to maintain all directinput controllers and whatever quirks they might have is not within the scope of the development team of this driver.

I will leave this issue open to dissuade duplicates and hear differing opinions.

cdes, 2016-05-26:
Well, I get what you're saying. It's the game dev's duty to support it.  However, we as end-users are pretty much screwed.

I don't understand anything related to drivers.  I was thinking that you might be able to hijack whatever the OS recognize and bind it to 360 driver.

My idea was like having a virtual 360 controller, that games recognize as a physical 360 controller, which is controlled by ANY pre-recognized device by the OS.

I certainly didn't mean that you should support all other direct input devices, I just asked if it's possible to make something like that.

FranticRain, 2016-05-26:
We can't just "highjack what the OS recognizes" because drivers are the thing that recognize these devices. With a driver, we either connect directly to the device, or that data is completely out of our control. 

Your idea of a virtual 360 controller would therefore require that we directly hook into every directinput controller in order to re-order their packets to match those of an Xbox controller, and then fake the VID and PID output. However, like I said before, this would mean supporting various different interfaces that we do not support such and bluetooth and wifi direct. It is impossible for use to say "Oh well we will just do this for the DS3 and DS4" because suddenly we have absolutely no ground to stand on when anyone else comes along and wants their device supported. Suddenly, we're supporting hundreds of directinput controllers, trying to manage that nightmare, while the actual Xbox controllers suffer because of it.

Once again, this is highly out of the scope of this driver, and I will under no circumstances support it. This is not a feasible concept and would hurt the development of the actual part of this driver that matters.

d235j, 2016-05-26:
There is software available such as [USB Overdrive](http://www.usboverdrive.com/USBOverdrive/News.html) which allows for rebinding buttons with arbitrary devices and whatnot. Check that out — your request is out of scope for this project.

closed 2016-05-26 by d235j